meshtastic/python #399: [Bug]: CLI Configure command not working

tmarkson, 2022-11-18:


tmarkson, 2022-11-18:
Sorry, I did.  The `node.conf` file is the output from this command that was run first on Windows: `meshtastic --export-config > node.conf`

caveman99, 2022-11-18:
please provide the output of `meshtastic --version`

if it is below 2.0.3, run `pip install --upgrade meshtastic`

tmarkson, 2022-11-18:
Thanks @caveman99 
The config option allows uploading a config file after upgrading the CLI to 2.0.3

Now in my limited usage of this command, the node will reboot before taking all settings.
Running the command a few times did not completely change the settings on the node either.
I think it's not ideal because a fresh node would not be configured completely 
Is this expected and/or normal? 

For example I do these steps:
1. upload latest firmware to a node
2. connect to PC via serial
3. upgrade latest CLI
4. export the config `meshtastic --export-config > config.yaml`
5. open `config.yaml`, change some items at different locations, such as `lsSecs` and `screenOnSecs`
6. run command `meshtastic --configure config.yaml`
7. run command to compare values `meshtastic --get power.ls_secs` and see it has original value.

tmarkson, 2022-11-18:
Tried to change a single config item via `--configure` command does not seem to change the value on node. 

The CLI command does run with a single item, so that's cool! 

Here are the contents of that file for test: 
```
# start of Meshtastic configure yaml
config:
  power:
    lsSecs: 301
```

garthvh, 2022-11-20:
I believe lsSecs is hard coded on the device at 300, setting it won't ever change it.

garthvh, 2022-11-20:
The firmware does reboot automatically for virtually every section of the config now, so this yaml configuration import does not work for all the settings at once any more.

tmarkson, 2022-11-20:
Thanks for the info. 

> I believe lsSecs is hard coded on the device at 300, setting it won't ever change it.

You're right,  the value did not change on the node after commanding a change with `--set`

Was there a reason for that?  Is that temporary behavior? 

> The firmware does reboot automatically for virtually every section of the config now, so this yaml configuration import does not work for all the settings at once any more.

That's unfortunate; is there a faster way to configure nodes than waiting for reboot after each setting?

closed 2022-11-21 by thebentern