meshtastic/python #410: meshtastic --tunnel

markbirss, 2022-12-17:
I am unable to establish a tunnel, gives error

`INFO file:tunnel.py __init__ line:83 Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO file:tunnel.py __init__ line:93 Node !818c6ff3 has IP address None.111.243
INFO file:tunnel.py __init__ line:93 Node !615f8e43 has IP address None.142.67
INFO file:tunnel.py __init__ line:93 Node !1cea1a76 has IP address None.26.118
DEBUG file:tunnel.py __init__ line:95 creating TUN device with MTU=200
None.111.243: Unknown host
ifconfig: `--help' gives usage information.
Aborting due to: ifconfig command failed.
DEBUG file:serial_interface.py close line:65 Closing Serial stream
DEBUG file:stream_interface.py close line:119 Closing stream
DEBUG file:stream_interface.py _sendToRadioImpl line:109 Sending: disconnect: true
DEBUG file:stream_interface.py _sendToRadioImpl line:114 sending header:b'\x94\xc3\x00\x02' b:b' \x01'
DEBUG file:stream_interface.py __reader line:186 reader is exiting
DEBUG file:stream_interface.py _disconnected line:80 Closing our port
`

garthvh, 2022-12-17:
This feature is super fragile

garthvh, 2023-03-19:
Can you try this again Mark? The whole TCP stack seems much more stable now.

luzpaz, 2023-07-28:
@markbirss :point_up_2:

markbirss, 2023-07-28:
> @markbirss point_up_2

same error for me still

`meshtastic --tunnel
Connected to radio
INFO file:tunnel.py __init__ line:83 Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO file:tunnel.py __init__ line:95 Node !25a78c24 has IP address None.140.36
INFO file:tunnel.py __init__ line:95 Node !25a78d38 has IP address None.141.56
None.140.36: Unknown host
ifconfig: `--help' gives usage information.
Aborting due to: ifconfig command failed.
^CTraceback (most recent call last):
  File "/usr/local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 1335, in main
    common()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 967, in common
    time.sleep(1000)`

jsburch83, 2023-09-22:
I've got net-tools installed on both boxes, and I'm getting the same error, which points to ifconfig command failing. 

meshtastic --tunnel
Connected to radio
INFO file:tunnel.py __init__ line:83 Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO file:tunnel.py __init__ line:95 Node !dadfb700 has IP address None.183.0
INFO file:tunnel.py __init__ line:95 Node !dadfb48c has IP address None.180.140
None.183.0: Unknown host
ifconfig: `--help' gives usage information.
Aborting due to: ifconfig command failed.

jsburch83, 2023-09-22:
I found that if you use the subnet command it works - 

meshtastic --subnet 10.200 --tunnel

markbirss, 2023-09-23:
@jsburch83 

2 x SX1280 2.4Ghz LoRa set SHORT_FAST

50 Pinss has 48% packet loss
[image]

Iperf3 has Retr
[image]

jsburch83, 2023-09-23:
@markbirss, so using the subnet command got it working?

markbirss, 2023-09-23:
> @markbirss, so using the subnet command got it working?

yes, but i previously modified the tunnel.py directly just to get it running

jsburch83, 2023-09-23:
@markbirss can you post the code changes?

marek22k, 2023-11-16:
Hi, since I find it stupid to create issues for my two comments, I'll write them here:
1. [Line 128](https://github.com/meshtastic/python/blob/70bb58b8ce50e7fe3d797b794d0005cf09c77818/meshtastic/tunnel.py#128) looks like this check could be omitted in the future. However, I would say that you should not blindly trust your peers or in this case the sender of the packet. For example, a malicious peer could join the mesh and exploit the lack of verification.
2. Since I'm not very familiar with Python and may be reading the script incorrectly, this point may be wrong: As I understand the code, every IP packet is sent to the Tun interface. However, it would actually only be necessary to send IP packets to the Tun interface that are intended for us. Perhaps a check in the following form somewhere in the script would make sense:
```python
if packet["to"] != self.iface.myInfo.my_node_num: 
    logging.debug("Ignoring message that is not for us")
    forwardPacket(packet)
```

r-tierney, 2024-01-04:
Created a PR which fixes the subnet issue here:
https://github.com/meshtastic/python/pull/476

Found a few issues raised for this now

closed 2024-01-07 by thebentern

r-tierney, 2024-01-07:
This has been fixed in release version 2.2.17 to use the new tunnel feature simply update to this latest release
```bash
pip3 install --upgrade meshtastic
```