meshtastic/python #416: "Nodes in mesh" is not valid JSON

tobymurray, 2023-01-16:
Example:

```
$ meshtastic --port /dev/ttyUSB0 --info
<...snip...>
Nodes in mesh:  {'num': 2885175728, 'user': {'id': '<REDACTED>', 'longName': 'ttgo 1.1', 'shortName': 't1', 'macaddr': '24:62:ab:f8:49:b0', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': <REDACTED>, 'longitudeI': -<REDACTED>, 'altitude': <REDACTED>, 'time': 1673813417, 'latitude': <REDACTED>, 'longitude': <REDACTED>}, 'lastHeard': 1673813630, 'deviceMetrics': {}}  {'num': 546312913, 'user': {'id': '<REDACTED>', 'longName': 'RAK4631', 'shortName': 'RK46', 'macaddr': 'f1:fc:20:90:12:d1', 'hwModel': 'RAK4631'}, 'position': {'latitudeI': <REDACTED>, 'longitudeI': <REDACTED>, 'altitude': <REDACTED>, 'latitude': <REDACTED>, 'longitude': <REDACTED>}, 'snr': 11.25, 'lastHeard': 1673813406, 'deviceMetrics': {'batteryLevel': 60, 'voltage': 3.829}, 'lastReceived': {'from': 546312913, 'to': 4294967295, 'id': 1849607190, 'rxTime': 1673813406, 'rxSnr': 11.25, 'hopLimit': 3, 'rxRssi': -73, 'fromId': '!<REDACTED>', 'toId': '^all'}, 'hopLimit': 3}
<...snip...>
```

The rest of the value outputs of `--info` are valid JSON. Are the nodes intentionally invalid?

Two changes would have to be made around [here](https://github.com/meshtastic/python/blob/master/meshtastic/mesh_interface.py#L98), making the overall `nodes` an array and I'm guessing use something like the existing `MessageToJson` to format the node info itself as JSON (single quotes instead of double).

karamo, 2023-03-29:
**My solution is:**
In `mesh_interface.py` change **line 98** from
`nodes = nodes + f"  {stripnl(n2)}"`

to
```
singlenode = f"{stripnl(n2)}"
nodes = nodes + "  " + singlenode.replace(chr(39),chr(34))
```

mverch67, 2023-03-30:
> ```
singlenode = f"{stripnl(n2)}"
nodes = nodes + "  " + singlenode.replace(chr(39),chr(34))
```

This is not the best solution, as the resulting list of nodes is also not in JSON format and the replacement of all occurent ' by " will cause issues where ' is desired.

tobymurray, 2023-04-15:
Looks like this was resolved in #433

closed 2023-04-15 by tobymurray