meshtastic/python #418: [Feature Request]: Serial over Bluetooth

TilCreator, 2023-01-23:
### Platform

NRF52, ESP32

### Description

I would really like to user the CLI over Bluetooth, so that I can use my laptop to connect to a node without using some WiFi AP. After a rough search it seams possible to combine Bluetooth serial with nimble, but I don't know that much about BLE and especially the low energy part.
It also seams possible to add BLE handling to the CLI itself.

vchrizz, 2023-05-19:
+1 on that!
I would also like to use serial over bluetooth, to connect the [Meshtastic<>Matrix relay](https://github.com/geoffwhittington/meshtastic-matrix-relay/) to a RAK4631 (NRF52 based) device.
The idea in my case would be to use the RAK4631 with a solar panel and have it connected over bluetooth to the relay, so you are not limited to an USB cable.
But there are sure also other uses for that.

geoffwhittington, 2023-05-19:
Is the recommendation to implement a StreamInterface or MeshInterface subclass for Bluetooth support?

vchrizz, 2023-05-20:
Maybe related, while looking for a solution I found this [hint on StackExchange](https://unix.stackexchange.com/questions/352494/alternative-to-the-now-deprecated-rfcomm-binary-in-bluez) about using python for serial communication:
```
$python3
>>> import socket
>>> sock = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_STREAM, socket.BTPROTO_RFCOMM)
>>> adapter = '00:11:22:33:44:55' #<adapter address>
>>> device = '55:44:33:22:11:00' #<device address>
>>> sock.bind((adapter, 1))
>>> sock.connect((device, 1))
## If not pinned it will ask you. You can use/adapt the bluez simple-agent for headless pinning
>>> sock.send(b'hello\n')
>>> sock.recv(100)
>>> sock.close()

rc14193, 2024-02-25:
Has this been addressed with the addition of https://github.com/meshtastic/python/blob/master/meshtastic/ble_interface.py
or is there more information on what needs to be added to that interface?

ianmcorvidae, 2024-06-02:
I think that this is fixed with the addition of the `bleak` based BLE handling, so I'll close this for now. However, if the creator of the issue has something else in mind I'd be fine to reopen it. It should be possible to use the CLI over bluetooth now, in any case.

closed 2024-06-02 by ianmcorvidae