meshtastic/python #422: Unclear licensing

michaellass, 2023-02-28:
The [LICENSE.txt](https://github.com/meshtastic/python/blob/19b4cd65ce8822bd236d2959ab61e1269df137fe/LICENSE.txt) in this repository contains the Apache 2.0 license and hence this is what GitHub shows as license for this package. However, the license specified in [setup.py](https://github.com/meshtastic/python/blob/19b4cd65ce8822bd236d2959ab61e1269df137fe/setup.py#L22) has been changed to GPLv3 in https://github.com/meshtastic/python/pull/392. Before that, it was specified as MIT in setup.py.

So, what is the current license of this code? If GPLv3, could the LICENSE.TXT be updated? If Apache, setup.py should reflect that as well.

garthvh, 2023-03-01:
It is now GPLv3 and was originally Apache I don’t think any work was done to do a formal license change so it is probably dual licensed at this point.

luzpaz, 2023-07-26:
Ticket can be closed

closed 2023-07-26 by garthvh