meshtastic/python #423: remote admin `--info` returns empty "Preferences" and "Module preferences"

andrekir, 2023-03-13:
```
(venv) meshtastic-python % meshtastic --dest '!remote_id' --info
Connected to radio

Requesting channel 0 info from remote node (this could take a while)
Requesting channel 1 info from remote node (this could take a while)
Requesting channel 2 info from remote node (this could take a while)
Requesting channel 3 info from remote node (this could take a while)
Requesting channel 4 info from remote node (this could take a while)
Requesting channel 5 info from remote node (this could take a while)
Preferences: {}

Module preferences: {}

Channels:
  PRIMARY psk=secret { "psk": "blablabla", "name": "ChannelName" }

Primary channel URL: https://meshtastic.org/e/#blablabla
```

andrekir, 2023-03-13:
reference https://github.com/meshtastic/python/commit/c55b1188e84bc12505975c41b1f8ea8308259b58#diff-a76fd12bdd151a069d544c7b36bd277ad10d518527c40c5929e8aecac5a7ca06L255-L314

GUVWAF, 2023-03-15:
Ideally the way how Python handles admin packets should be rewritten completely, because it now has a local representation of all its settings and it uses this to generate the updated admin packets. Therefore, before doing any admin command it gets all channels and all settings, but the latter was removed, so the result is empty. 

I'm wondering if the C# CLI is mature enough to promote using that, I will need to test if remote admin works there already. If it does, maybe we just need to remove remote admin from Python.

Indeed, `--info` and `--nodes` may be something that's not really suited for this, because it will always need many packets.
However, we tested a burst of packets and that is now working almost perfect. With two nodes in the mesh, it's hard to get any failures now, with more than that, it only sometimes misses the real ACK. Channel utilization indeed goes up quickly, but since it's only for a short time, I don't think it's really a problem unless you have a big mesh.

PR #428 (2023-03-19): Get and set methods for remote node using one admin message
Now the commands `--get` and `--set` work for remote nodes by using only one admin message per field and it doesn’t request all channels. When setting, it first requests the current config of the field, then sends an updated admin message, after which it waits for an ACK from the remote node.

This partly fixes #423. When trying `--info` for a remote node, it prints that you need to use `--get` with a specific field. Also, it prints that `--nodes` for a remote node is not supported (currently it was showing the list of the connected n [... cut]
closed 2023-03-19 by thebentern