meshtastic/python #424: update admin channel naming convention

andrekir, 2023-03-13:
convention was updated to accept "admin" in any case format (ie. "Admin", "admin", "ADMIN", etc) but python CLI still requires all lowercase "admin" channel name.

ammarfitwalla, 2023-04-20:
Hi, can I work on this issue?

garthvh, 2023-04-20:
Sure

pranshugupta01, 2023-09-07:
Hi , can i work on this issue?

andrekir, 2023-09-07:
sure. see: https://meshtastic.org/docs/configuration/remote-admin

TimothyHarder, 2024-01-07:
Hi guys, I'm just starting to learn the project so please forgive my simple question. 

This seems like a pretty simple change, just calling lower in the [ch_add block after the length check](https://github.com/meshtastic/python/blob/6dcdf7fc19d2c0e30a97c415e4ab0b9105b1b2a4/meshtastic/__main__.py#L599), or in the [getChannelByName function](https://github.com/meshtastic/python/blob/6dcdf7fc19d2c0e30a97c415e4ab0b9105b1b2a4/meshtastic/node.py#L253). Do you all want all channel names cast to lowercase, or is just the admin channel case-insensitive?

andrekir, 2024-01-07:
[_getAdminChannelIndex](https://github.com/meshtastic/python/blob/6dcdf7fc19d2c0e30a97c415e4ab0b9105b1b2a4/meshtastic/node.py#L267) needs to find any case-insensitive variant of "admin" like: Admin ADMIN ADmin etc

maybe something like:

```
def _getAdminChannelIndex(self):
    """Return the channel number of the admin channel, or 0 if no reserved channel"""
    for c in self.channels or []:
        if c.settings and c.settings.name.lower() == "admin":
            return c.index
    return 0
```

TimothyHarder, 2024-01-07:
Definitely a good idea, though I think modifying user input has merit, or we could end up with a channel list like so:
["admin", "Admin", "ADMIN"] and _getAdminChannelIndex would return the first of them. I was thinking the self.channels could be all case insensitive, all added channels would be stored lowercase. Admittedly though, I don't know the project very well yet and wouldn't know the implications of making a change like that.

That said, I've got a PR in with your suggested change for now! #478

closed 2024-01-16 by thebentern