meshtastic/python #438: I would prefer to keep the ID in JSON

karamo, 2023-04-01:
As a result of fix https://github.com/meshtastic/python/pull/433

I would prefer to keep the ID in JSON in "**Nodes in mesh:**" in the output of `--info`

In `mesh_interface.py`
in `def showInfo(self, file=sys.stdout):`

delete the line:
`               n2['user'].pop('id')`

mverch67, 2023-04-02:
Programmatically this request doesn't really make sense, as the id is already part of the dictionary key and can be easily obtained in any programming/scripting language with json-parser, e.g:

$ jq 'keys[] as $k | "\($k)=\(.[$k] | .user.longName)"' nodes.json
"!55c58b64=Meshtastic-8b64"
"!fa5a0688=Meshtastic-0688"
"!fa5a06bc=Meshtastic 06bc"
"!fa5a06f8=Meshtastic 06f8"

karamo, 2023-04-02:
In my understanding of data structures, i.e. KEY / VALUE pairs, the key is **always constant and never variable.**

The **KEYS** are for example: "**node**", "**name**", "**id**", "**num**" …
but never a variable as **"!1234abcd"**

garthvh, 2023-04-02:
> e for example: "**node**", "**name**", "**id**", "**num**" …
> but never a variable as **"!1234abcd"**

The "!1234abcd" is the user.id value, it is derived by removing the first 4 characters from the MAC address

mverch67, 2023-04-02:
The node id is always constant because it is derived from the MAC address. It will never change. This is used as a key in the dictionary (of nodes).

closed 2023-12-16 by garthvh