meshtastic/python #444: Bug: TypeError: 'NoneType' object is not subscriptable

mverch67, 2023-04-11:
Sometimes, when receiving a packet (which cannot be decoded) while the meshtastic command is running, we get an exception:

```
ERROR file:stream_interface.py __reader line:175 Error while handling message from radio 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/home/manuel/.platformio/penv/lib64/python3.11/site-packages/meshtastic/stream_interface.py", line 173, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/home/manuel/.platformio/penv/lib64/python3.11/site-packages/meshtastic/mesh_interface.py", line 745, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/home/manuel/.platformio/penv/lib64/python3.11/site-packages/meshtastic/mesh_interface.py", line 916, in _handlePacketFromRadio
    portnum = decoded["portnum"]
              ~~~~~~~^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

luzpaz, 2023-07-26:
@mverch67 is this ticket solved ?

mverch67, 2023-07-27:
> @mverch67 is this ticket solved ?

yes, in #445, see the mentioning above your post

luzpaz, 2023-07-28:
Please close ticket in that case, thanks

closed 2023-07-28 by thebentern