meshtastic/python #448: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

jeremiah-k, 2023-06-04:
I was hopeful for the fix in #445, as the NoneType errors were repeating in the logs when using the [matrix relay](https://github.com/geoffwhittington/meshtastic-matrix-relay), which makes use of this API.

The good news is that the 'NoneType' errors are gone, but unfortunately they've been replaced with this error and it occurs at the same frequency:
```
ERROR:root:Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/home/jeremiah/dev/meshtastic-matrix-relay/.pyenv/lib/python3.10/site-packages/meshtastic/util.py", line 232, in _run
    o()
  File "/home/jeremiah/dev/meshtastic-matrix-relay/.pyenv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 958, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/jeremiah/dev/meshtastic-matrix-relay/.pyenv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/jeremiah/dev/meshtastic-matrix-relay/.pyenv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/jeremiah/dev/meshtastic-matrix-relay/.pyenv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface
```

As far as I can tell neither set of errors affect its functionality, but they're always present in the logs.

mverch67, 2023-07-27:
I don't have matrix-relay installed for tests, but what is the output of the logging command in line 956? It should reveal the source of the error in 958:

```
        logging.debug(f"Publishing {topic}: packet={stripnl(asDict)} ")
        publishingThread.queueWork(
            lambda: pub.sendMessage(topic, packet=asDict, interface=self)
        )
```

jeremiah-k, 2023-10-19:
> I don't have matrix-relay installed for tests, but what is the output of the logging command in line 956? It may reveal the source of the error in 958:
> 
> ```
>         logging.debug(f"Publishing {topic}: packet={stripnl(asDict)} ")
>         publishingThread.queueWork(
>             lambda: pub.sendMessage(topic, packet=asDict, interface=self)
>         )
> ```
> 
> What happens if you remove the "interface=self" parameter in line 958?

I'm sorry I missed your reply to this.  I've been running 2.2.5 in a venv w/ mmrelay for a few weeks and realized that the errors are gone & was coming back to close this issue.  So it seems to have been working since then at least.

Thank you.

closed 2023-10-19 by jeremiah-k

migillett, 2025-01-31:
Sorry to re-open this thread, but I've been having issues with this lately. I'm using this library as an interface for a custom-built BBS. My traceback looks like this:
```
2025-01-30 21:37:50 Eastern Standard Time - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "\GitHub Projects\Meshtastic-Listener\venv\Lib\site-packages\meshtastic\util.py", line 309, in _run
    o()
    ~^^
  File "\GitHub Projects\Meshtastic-Listener\venv\Lib\site-packages\meshtastic\mesh_interface.py", line 1541, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\GitHub Projects\Meshtastic-Listener\venv\Lib\site-packages\pubsub\core\publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^
  File "\GitHub Projects\Meshtastic-Listener\venv\Lib\site-packages\pubsub\core\topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "\Meshtastic-Listener\venv\Lib\site-packages\pubsub\core\topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
                                    list(msgData.keys()), optional - set(self.allOptional))
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface
```

I've taken the `interface=self` out of my forked copy of this repo and have not any traceback errors nor have I had any lapse in features. I've made a branch and a PR for this. https://github.com/meshtastic/python/pull/733

migillett, 2025-02-01:
Just realized my error. For those who come here afterwards, make sure whatever function you write that's triggered by the pubsub topic `meshtastic.receive` has 3 params:

```
def __on_receive__(packet, interface, node_list) -> None:
```
the packet is a dictionary, interface is a `MeshInterface` class and node_list is a list.