meshtastic/python #452: How to measure the actual lora bandwidth

wengjiahuang0529, 2023-07-13:
I have successfully installed Meshtastic devices using LoRa communication on two Raspberry Pi 3.

I want to implement a tool to measure the bandwidth of a LoRa connection by observing the dispersion time of two consecutive icmp packets(back-to-back) sent to the destination. (i.e. similar to CapProbe method，http://slideplayer.com/slide/4827053/)

Can you recommend a suitable tool or API to enable two LoRa devices to exchange packets for bandwidth measurement ? Alternatively, if there are any existing tools dedicated to measuring the actual bandwidth of a LoRa connection, your suggestions would be greatly appreciated.
Thanks.

garthvh, 2023-07-13:
The bandwidth is a part of the lora config.  Each modem preset has a max bandwidth value.  Meshtastic uses flood routing so using TCP networking tools is not going to work.

wengjiahuang0529, 2023-07-13:
> 
Ok, I still have some questions . 
I have two lora devices, is it possible to flood two back-to-back packets from one device to the other device through lora ?

closed 2023-12-16 by garthvh