meshtastic/python #456: api 2.2.0 talking with fmwr 2.2.0 produces error opening interface

vinloren, 2023-08-11:
Got Exception: This version of meshtastic-python requires device firmware version 1.2 or later. For more information see 
https://tinyurl.com/5bjsxu32 upon statement interface = meshtastic.serial_interface.SerialInterface(). 
``
import meshtastic
import meshtastic.serial_interface
from pubsub import pub

def onReceive(packet, interface): # called when a packet arrives
    print(f"Received: {packet}")

def onConnection(interface, topic=pub.AUTO_TOPIC): # called when we (re)connect to the radio
    # defaults to broadcast, specify a destination ID if you wish
    interface.sendText("hello mesh")

pub.subscribe(onReceive, "meshtastic.receive")
pub.subscribe(onConnection, "meshtastic.connection.established")

geeksville, 2023-08-11:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/device-firmware-2-2-0-python-api-2-2-0-error-on-meshtastic-serial-interface-serialinterface/8001/3

GUVWAF, 2023-08-11:
Are you sure that the imported version of meshtastic in your script is also on 2.2.0, and not only the CLI? Maybe you have multiple versions of the package on your system. Since this check was already removed from the source code here: https://github.com/meshtastic/python/commit/055da95b8aa10049b50e03e14ac09cf8a56e7dbc 

What does this show?
```python
import pkg_resources
print(pkg_resources.get_distribution("meshtastic").version)
```

vinloren, 2023-08-11:
Sorry, it was my fault, actually meshtastic was updated to 2.2.0 on a different python ver (I have 2 versions on my PC) but the one driving my application. After updating meshtastic on the correct python version now all is fine.

closed 2023-08-11 by vinloren