meshtastic/python #465: [Feature request] Possibility to specify a channel for IP tunneling

marek22k, 2023-11-23:
Hello,

I've seen that Meshtastic also offers the possibility to tunnel IP packets, which I think is great.
My question is: Which channel does Meshtastic use by default?
Based on this, my feature request is: I would like to have a command line flag that allows you to select the channel you use to set up the IP tunnel.

caveman99, 2023-11-23:
Primary channel would be standard for broadcast packets.

RicInNewMexico, 2023-11-23:
The IP tunnel is pretty much unusable unless you happen to be running very fast modem configuration. Even then it has VERY high latency. Also many countries have a limit on channel utilization / duty cycle which will prevent the tunnel from maintaining a connection. 

Meshtastic just isn't well suited to IP traffic.

marek22k, 2023-11-23:
> Primary channel would be standard for broadcast packets.

I see. I actually wanted to create a separate channel for my tests with the IP tunnel (which I could then also encrypt with a random PSK).

mc-hamster, 2023-11-23:
> > Primary channel would be standard for broadcast packets.
> 
> 
> 
> I see. I actually wanted to create a separate channel for my tests with the IP tunnel (which I could then also encrypt with a random PSK). However, using the extra tunnel as the primary tunnel would also mean that other information such as NodeInfos would also only be sent there, as far as I understand it.

We keep traffic separate by using ports. See:

https://github.com/meshtastic/protobufs/blob/master/meshtastic/portnums.proto

closed 2023-11-24 by garthvh

marek22k, 2023-11-24:
@garthvh For my understanding: The reason for closing is that IP Tunnel uses broadcast, and broadcast only works on the primary channel?

garthvh, 2023-11-24:
> @garthvh For my understanding: The reason for closing is that IP Tunnel uses broadcast, and broadcast only works on the primary channel? 

I would start by getting it working, I see you found the open issue.  The tunnel is a POC, not a real usable TCP feature.

marek22k, 2023-11-24:
So the point of being able to use broadcast on other channels is already in the other issue? I must have overlooked that.