meshtastic/python #467: Tunnel: ifconfig are outdated

marek22k, 2023-12-04:
Hello,

ifconfig is obsolete on Linux and should be replaced by ip. On some Linux distributions, ifconfig is no longer (pre)installed.
```
$ sudo meshtastic --tunnel
Connected to radio
INFO file:tunnel.py __init__ line:83 Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO file:tunnel.py __init__ line:95 Node !5687b499 has IP address None.180.153
INFO file:tunnel.py __init__ line:95 Node !148bfdc5 has IP address None.253.197
sh: line 1: ifconfig: command not found
Aborting due to: 
```

geeksville, 2023-12-05:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/meshtastic-tunnel-sh-1-ifconfig-not-found/3818/4

OpenSource-For-Freedom, 2023-12-15:
run: ip addr, some unix distro's require that. 

Also have you installed "net-tools"

$ sudo apt-get install net-tools

marek22k, 2023-12-16:
ifconfig is deprecated (see above) and should no longer be used on Linux systems. Therefore `ip` should be used.

Yes, installing `net-tools` solves the symptoms, but not the actual problem.

r-tierney, 2024-01-05:
Looking at the code in tunnel.py

it looks like its using the pytap2 library here
https://github.com/meshtastic/python/blob/master/meshtastic/tunnel.py#L23

ifconfig from the pytap2 library is used here:
https://github.com/meshtastic/python/blob/master/meshtastic/tunnel.py#L115

Looking at the pytap2 homepage it states a dependency on `net-tools`
https://github.com/johnthagen/pytap2

I believe this third party dependency should probably be included in the Meshtastic documentation.

r-tierney, 2024-01-05:
Looking at the code in tunnel.py

it looks like its using the pytap2 library here
https://github.com/meshtastic/python/blob/master/meshtastic/tunnel.py#L23

ifconfig from the pytap2 library is used here:
https://github.com/meshtastic/python/blob/master/meshtastic/tunnel.py#L115

Looking at the pytap2 homepage it states a dependency on `net-tools`
https://github.com/johnthagen/pytap2

I believe this third party dependency for `net-tools` should probably be included in the Meshtastic documentation.

could also raise this deprecation of ifconfig with the pytap2 library maintainers and get this fixed upstream

However looking at this comment it looks like they are well aware of this deprecation:
https://github.com/johnthagen/pytap2/blob/master/src/pytap2/__init__.py#L162
```python
        # TODO: New systems like Ubuntu 17.04 do not come with ifconfig pre-installed.
        ifconfig_cmd = "ifconfig {} ".format(self._name)
```

garthvh, 2024-03-16:
Is this still out of date?

marek22k, 2024-03-16:
Yes, it is legacy on Linux (i.e. not BSD).

r-tierney, 2024-03-17:
Just bumped the PR I made on the pytap2 repo to make iproute2 the default instead of ifconfig