meshtastic/python #475: Consistent Data Retrieval

kb3mpn, 2024-01-03:
[image]

In the former command, I'm able to see the GPS data for the extant node. In the latter, I get a simple text version of some of the data. For the purposes of grep/sed/awk, it would be lovely to be able to poll a meshtastic node and get a common response -- preferably one without ascii tables to fight over.

It would be nice to have a python library that manages this, or the ability to pull the information down in JSON from an API endpoint.

Thank you for all of the work that you're doing on Meshtastic, it's fun as hell and I look forward to playing with it a lot this year.

ethzero, 2024-03-15:
I heartily +1 this.

Currently I can only crudely dump this information into a webpage as-is (wrapping it with <pre> tags).

It would be fantastic to have JSON output for *all* Meshtastic commands much in the way nmap and the like have an output type flag for JSON,XML,YAML, etc.

This would also help solve the issue that the --info command squirts out JSON-like strings but is otherwise un-parsable as JSON because it's peppered with plain-text.

garthvh, 2024-03-16:
There is not really anyone maintaining the CLI, the best approach is to use the python API https://python.meshtastic.org

ianmcorvidae, 2024-05-09:
This is a great thing that I'd like to do, now that I'm the maintainer for this (i.e., we have one again)! I've made #569 as an issue to discuss the data format we might want (and will be making other issues for discussion of things like CLI options & subcommands, which is related). I'll close this to direct discussion there, and input would be appreciated!

closed 2024-05-09 by ianmcorvidae