meshtastic/python #480: When using standalone CLI, --nodes gives: No module named 'timeago.locales.en'

Scarg, 2024-01-15:
## Description 
When using the standalone CLI (tested both Ubuntu and Windows) while things like `--info` do work, running `meshtastic --nodes` results in the following error:

```bash
Connected to radio
Aborting due to: No module named 'timeago.locales.en'
```

This was already reported some time ago here: #331.

**It works fine if you do not use the standalone version, to reproduce the issue the standalone version must be used.**

Considering that the ticket was closed too quickly last time, I'll try to write down a "how-to replicate" section.

## Reproduction steps
### Downloaded version
1. Download https://github.com/meshtastic/python/releases/download/2.2.17/meshtastic_ubuntu or https://github.com/meshtastic/python/releases/download/2.2.17/meshtastic_windows.
2. Execute the downloaded file with the `--info` parameter (via USB or via `--host ipAddr` it is the same), the result should be fine
3. Execute the downloaded file with the `--nodes` parameter (via USB or via `--host ipAddr` it is the same) the result will be
`Aborting due to: No module named 'timeago.locales.en'`

### Build it
1. As per the [build-and-publish-windows](https://github.com/meshtastic/python/blob/2eda2e56d5781a3af365ce43411e3697f01b4feb/.github/workflows/release.yml#L180-L183) workflow, run in the repository:
1.1 `pip uninstall meshtastic #at least i guess so, I have no clue`
1.2 `pip install pyinstaller`
1.3 `pip install -r requirements.txt`
1.4 `pip install .`
1.5 `pyinstaller -F -n meshtastic --collect-all meshtastic meshtastic/__main__.py`
2. Now run `.\dist\meshtastic.exe --nodes` (windows) or `.\dist\meshtastic.exe --nodes` (ubuntu) 
3. The result should be `Aborting due to: No module named 'timeago.locales.en'`

If step 1.5 is slightly changed to `pyinstaller -F -n meshtastic --collect-all meshtastic --add-data "%USERPROFILE%\AppData\Local\Programs\Python\Python310\Lib\site-packages\timeago\locales\en.py;timeago/locales" meshtastic/__main__.py` (on windows at least) the built version works fine.

## Notes
I have no idea of what I'm doing 
Also might be related to https://github.com/hustcc/timeago/issues/40#issue-1116686280 "Locales not imported with PyInstaller".

mattriney, 2024-01-29:
This does seem to be related to the issue #40  listed above for "timeago."

I just ran into the same issue with 2.2.19 version of the stasndalone cli:

I did not built it, I only downloaded from the releases.

Exact same error message:

Aborting due to: No module named 'timeago.locales.en'

dgarley, 2024-03-10:
Same here with downloaded Win standalone 2.2.22, reproduceable  as per OP Scarg  steps in "Downloaded version" above.  
Running on:
 OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045

garthvh, 2024-03-16:
The standalone windows version has been removed from distribution

closed 2024-03-16 by garthvh