meshtastic/python #481: --reply sends response to channel, not to sender

rlmalisz, 2024-01-21:
--reply responds to direct messages.  It seems wrong for those responses to go to the entire channel the DM came in, rather than just to the sender.  On a private channel, this is pretty annoying.  On the primary/default, it's needless channel pollution.

andrekir, 2024-01-23:
keep in mind direct messages didn't exist when this reply code was written. can you give more context on the use case or what you're trying to do?

rlmalisz, 2024-01-23:
> 
> 
> keep in mind direct messages didn't exist when this reply code was written. can you give more context on the use case or what you're trying to do?
I have a headless rPICO that I am playing with.  Assuming I can put the pieces together, it’s destined to be placed somewhere high (up in a tree, hanging from a balloon, on a drone) as a temporary site relay.  None of those use cases argue for a display, so CLI is the easiest way to verify connectivity and function.

I have to confess that I hadn’t known that —reply would reply to channel messages…in response to my query about it on python#discourse, Garth said it replied to DMs.  I took that as gospel, but having looked at the code, should have realized that it simply responds to any inbound message, and replies on the channel it came in on.

This isn’t a huge problem for me, as I can do my testing on a private channel.  But I can see the utility in having two flavors: —reply-all and —reply-sender.  For use on the default/primary, the latter could be used to avoid producing a lot of channel-wide chatter to innocent bystanders.  Better naming might be “—reply” and “—reply-all”, to give the user a hint that they’re broadcasting.

In a dream world, I’d love to see a device setting to put a node into —reply-sender mode until unset.  But that would be a firmware job.

Looking at  the code, I didn’t see where you’d retrieve the sender’s node address, but am unfamiliar with some of the data constructs being used.

Does any of that make sense?

—Richard

andrekir, 2024-01-24:
reply is meant for testing, nothing fancy.

perhaps there are similar options baked in the firmware you can use, from position broadcasts to traceroutes.

if you like python check: https://github.com/datagod/meshwatch

it's not updated for 2.0, but can give you some ideas on how to add more functionality using the API.

rlmalisz, 2024-01-24:
I made a very modest change to __main__.py on my machine:

def onReceive(packet, interface):
    """Callback invoked when a packet arrives"""
    our_globals = Globals.getInstance()
    args = our_globals.get_args()
    try:
        d = packet.get("decoded")
        logging.debug(f"in onReceive() d:{d}")

        # Exit once we receive a reply                                                                                      
        if (
            args
            and args.sendtext
            and packet["to"] == interface.myInfo.my_node_num
            and d["portnum"] == portnums_pb2.PortNum.TEXT_MESSAGE_APP
        ):
            interface.close()  # after running command then exit                                                            

        # Reply to every received message with some stats                                                                   
        if args and args.reply:
            msg = d.get("text")
            if msg:
                rxSnr = packet["rxSnr"]
                rxRssi = packet["rxRssi"]
                hopLimit = packet["hopLimit"]
		sender = BROADCAST_ADDR
                if packet["to"] == interface.myInfo.my_node_num:
		    sender = packet["from"]
                print(f"message: {msg}")
                reply = f"got msg '{msg}' with rxSnr: {rxSnr} rxRssi: {rxRssi} and hopLimit: {hopLimit}"
                print("Sending reply: ", reply)
                interface.sendText(reply,sender)

    except Exception as ex:
        print(f"Warning: There is no field {ex} in the packet.”)

I was interested in RSSI values as well…part of what I am doing is trying to site a relay, so signal strength numbers are of interest to me, so I added them.  But aside from that, the code changes reply to DMs with DMs, channel messages with channel messges.

—Richard

garthvh, 2024-03-16:
Reply module was removed

closed 2024-03-16 by garthvh