meshtastic/python #491: meshtastic --ble stuck after "Connected to radio"

dirkbeer, 2024-02-25:
Hi - Really appreciate the BLE support!

I'm having a problem where  `meshtastic --ble` stops after connecting to the radio and nothing further happens:

```
root@cottonmouth:~# . meshtastic/bin/activate
(meshtastic) root@cottonmouth:~# bluetoothctl devices
Device F7:BE:57:1F:90:05 dto_9005
(meshtastic) root@cottonmouth:~# meshtastic --ble-scan
Found: name='dto_9005' address='F7:BE:57:1F:90:05'
BLE scan finished
(meshtastic) root@cottonmouth:~# meshtastic --ble dto_9005
Connected to radio

```

Once I Ctrl-C out of that, `meshtastic --ble-scan` can no longer see the device:

```
(meshtastic) root@cottonmouth:~# meshtastic --ble F7:BE:57:1F:90:05
Connected to radio
^CException ignored in: <module 'threading' from '/usr/lib/python3.11/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1583, in _shutdown
    lock.acquire()
KeyboardInterrupt:
(meshtastic) root@cottonmouth:~# meshtastic --ble-scan
BLE scan finished
(meshtastic) root@cottonmouth:~# bluetoothctl devices
Device F7:BE:57:1F:90:05 dto_9005
```

Here's my info:

meshtastic --support
```
 System: Linux
   Platform: Linux-6.1.0-rpi8-rpi-v8-aarch64-with-glibc2.36
   Release: 6.1.0-rpi8-rpi-v8
   Machine: aarch64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v2.2.22
 Executable: /root/meshtastic/bin/meshtastic
 Python: 3.11.2 CPython GCC 12.2.0
```
meshtastic --info
```
Warning: No Meshtastic devices detected.
```

dirkbeer, 2024-02-25:
Looks like I was just using it wrong, didn't realize that the `--ble dto_9005` has to be included with every command. For example,

`meshtastic --ble dto_9005 --nodes`

works just fine.

ianmcorvidae, 2024-06-08:
So yeah, the `--ble` bit does need to be included with every command. If you're still seeing it hung after `Connected to radio`, you could try running with `--debug`, which will at least show if it's still attempting to do stuff or not. That might narrow down what's causing it to hang up, potentially.