meshtastic/python #493: Preferences settings sometimes not sticking / saving. CLI 2.2.22 Firmware 2.2.24 App 2.2.24

talkingnews, 2024-03-01:
I have 3 Heltec version 3 and I'm having random problems with settings not "sticking", both via command line and app.

meshtastic cli version 2.2.22
Android app version 2.2.24
Heltec v3 firmware version 2.2.24 (full clean install all on 3).

I found the (fairly old) issue below which seem to indicate that the problem has been resolved. But I'm still experiencing it.

"preference changes not saving in 1.1.8"
https://github.com/meshtastic/python/issues/41

I thought it was only CLI, but I just updated my fixed position, and it didn't stick via the app.

I have "cleaned" the command line input to ensure that there are no weird hidden unicode characters, or t0o many spaces, or silly things like 1 or 3 dashes before each "--set".

I had more luck by always setting the region as the first thing after a flash, then waiting and setting other things.

That seems the most reliable, but then I noticed that, having set my "short name", when I added a channel, it reset my short name to default.

This seems to be the same issue as experienced by a user on Discord:

> "Sometimes after saving a setting - like enabling the telemetry module for example, I’ve noticed that the module says it’s enabled but when i disconnect and reconnect the device I see the setting wasn’t saved. It doesn’t happen all the time but it’s happened enough for me that I always double check my saved settings."

So then I found this:

"1.3.40: CLI settings inputs break other settings? Just me?"
https://meshtastic.discourse.group/t/1-3-40-cli-settings-inputs-break-other-settings-just-me/6493

I've even tried grouping the settings by module, or setting each one individually and not chaining at all.
I'm using a good quality, new USB cable. I've even tried slowing the serial speed right down, but I don't think it's related to either of those because I always flash successfully at 921,600 and if it was data corruption, I'm sure a firmware update would feel it more than a few bytes of command line.

Of course, it's proving impossible to reliably reproduce. Was thinking of writing a small script which would repeatedly make 1 or more settings, wait a few seconds, then read it back, compare the two, and log any that didn't change (and run on a loop, say, 50 times doing a full re-flash every 10 times).

Would that be of any use, or is there a simpler way of doing it?

francisuk1989, 2024-03-21:
i have experienced this in [2.3.1.4fa7f5a Alpha](https://github.com/meshtastic/firmware/releases/tag/v2.3.1.4fa7f5a) with a heltec wireless tracker v1.1 what i recently bought so is all new.

Is mainly the fixed altitude (alt) thats an issue for me, even tired setting it in the android app it just stays there for a little bit and then pufff gone.

I have tired the python cli also but i dont think is even working, It seems to set ok.

No issues with setting the lon and lat thats seem to stayed there.

francisuk1989, 2024-03-22:
Update, I have downgraded to ` 2.2.24.e6a2c06 ` with a full wipe and now the alt is working fine,