meshtastic/python #494: [Feature Request]: Is it possible to add an antenna connection test?

clwgh, 2024-03-02:
I have a Heltec LoRa v3 board and I believe I've destroyed the ipex connector on the supplied mini antenna by exceeding the number of mating cycles over the last few days of testing. Under a microscope the center hole looks pushed down with soft worn plastic around it. I don't think it's making an electrical connection to the center pin on the board connector.

Is it possible for a combination of firmware ability and CLI interfacing to be able to ask the radio to detect if an antenna is present? For example send a low power signal and measure the VSWR, if that is something that various chips can do? And then access this capability via a command such as `meshtastic --antenna-test`.

Just an idea for thought if this is possible, given the hobbyist and testing nature of this project, and the proliferation of ipex connectors which are not designed for more than a few mating cycles.

OpenSource-For-Freedom, 2024-03-05:
Are you thinking a test similar to an SWR test? Otherwise to test and antenna you'll need another node to measure input/ output Rx Tx and measure output power using pyserial. 

---

What if we build a signal test software package that uses "SWR" signals under a platform that pushes a "pulse" using buffs(?) or LV signal to just test the cable? Similar to a LV Toner? Could look into the ESP32 libraries as well, may already be there. 

Or

Push the CLI "SWR" package to push a GUI on the desktop and show a simple: 

Import tkinter
Import pyserial

# some code that measures USB output power and measures against a stored table for reference 

try:
    while True:
        if ser.in_waiting > 0:
            line = ser.readline().decode('utf-8').rstrip()

# code supplies the output for the table 

print(f"Received data: {line}")

I believe pyserial can monitor its own output power, so the script would simple need a static table to measure current power output, and then have your cable length and antenna dbm rating to get the true math. 

Then, you could run 

chmod +x -i pyserial_request_show.py

"pyserial_request_show.py" 

Or similar file once the tables are built. 

This is not my strongest suite, but wouldn't be cool to have our own SWR as a service? 

"SWRiiS" 🧐

clwgh, 2024-03-05:
The use case I'm describing is a hobbyist gets their Meshtastic board (in my case a Heltec LoRa V3), connects the antenna and then runs the command to confirm that the radio thinks the antenna is attached. And the command returns either a "Looks like your antenna is attached, you're good to go" or "Your antenna does not appear to be attached". This will help the hobbyist be sure that their antenna is making a usable connection to the board, especially if it's been "playing up" or they've had suspciously empty airwaves.

I have no idea if the Meshtastic firmware is able to make this detection, and I'm assuming that it would need to measure something like VSWR, and that of course the various radio chipsets would need to support being able to present that data for Meshtastic to interrogate. Maybe some would and some wouldn't.

This GitHub issue is just me raising the idea in case that is indeed possible, as this would be a useful feature. However if it's not possible for Meshtastic to determine this, well nothing was lost by putting the idea forward.

OpenSource-For-Freedom, 2024-03-05:
All ideas are always welcome.