meshtastic/python #495: Not sending requested config change over remote admin device

isaacbentley, 2024-03-04:
**Description**
When using the standalone CLI  for Windows  and pip installed for linux using the `--dest !<device ID>
` it's not setting the configuration as expected.

**Reproduction steps**

Download https://github.com/meshtastic/python/releases/download/2.2.22/meshtastic_windows.
Execute the downloaded file with the --dest parameter (via USB), results in executing the commands such as `--setlat` and `--setlon` successfully but it cofingure the local device, not the remoe device as targeted in the `--dest` command. 

**Notes**
Verfied that I'm able to remote configure remote devices via IOS & Andriod applications.

isaacbentley, 2024-03-04:
Thank you for the quick response! I'll try to update the location by your suggestion and update this comment

garthvh, 2024-03-16:
Remote location configuration is not available as it would allow nodes to change each other's location

closed 2024-03-16 by garthvh