meshtastic/python #499: Latest Release of meshtastic_windows is flagged as Malware

AeroXuk, 2024-03-12:
Windows Defender detects the file as `Trojan:Win32/Wacatac.H!ml` and puts it in quarantine.

[image]

I exported the file from quarantine and uploaded it to virus total, which gives several warnings:

https://www.virustotal.com/gui/file/b04277aed69410f4fbb5d023520bad1a2952735c13ebaa94176fd79cc07d81ff/detection

These are the file hashes of the file:
```
MD5       a466354404e43a4b8514b001e90bd04a 
SHA-1     9d147dff53043b19fd291e97562febfd965983d7 
SHA-256   b04277aed69410f4fbb5d023520bad1a2952735c13ebaa94176fd79cc07d81ff
```

clwgh, 2024-03-13:
I am also seeing the same thing with a Windows 10 VM.

[image]

garthvh, 2024-03-15:
They are false positives, but the standalone exe I has no one maintaining it, so better to just remove it

clwgh, 2024-03-15:
Thanks for confirming it's false positives. Can you restore the 2.3.0 windows version please? I'll save that as the last good copy.

Kralg, 2024-03-19:
Hi, did you remove windows stand alone from all the release? Can somebody push it back?