meshtastic/python #501: Unable to read local device, not found in DB

Kralg, 2024-03-13:
Hello,
I try to read GPIO from python CLI but with command I get warning and no output: $ ./meshtastic --gpio-rd 0x2000 --dest 2860018834
Connected to radio
Reading GPIO mask 0x2000 from 2860018834
Warning: NodeId 2860018834 not found in DB

With  ./meshtastic --info
Owner: Meshtastic 6c92 (6c92)
My info: { "myNodeNum": 2860018834, "minAppVersion": 30200 }

Can you advice what can be wrong here?

Kralg, 2024-03-13:
Thanks both work.
Do you know how can I trigger the remoteHardware module handling commands sent over textMessages?

GUVWAF, 2024-03-13:
Unfortunately that's not possible.

Kralg, 2024-03-13:
Actually one more question about --gpio-rd. I think I read that it should return ON or OFF but I am getting and it seems the same values as mask, is that correct?
:
$ ./meshtastic.exe --gpio-rd 0x2000 --dest "^local"
Connected to radio
Reading GPIO mask 0x2000 from ^local
Received RemoteHardware type=READ_GPIOS_REPLY, gpio_value=8192 value=8192
$ ./meshtastic.exe --gpio-rd 0x100 --dest "^local"
Connected to radio
Reading GPIO mask 0x100 from ^local
Received RemoteHardware type=READ_GPIOS_REPLY, gpio_value=256 value=256
$ ./meshtastic.exe --gpio-rd 0x4 --dest "^local"
Connected to radio
Reading GPIO mask 0x4 from ^local
Received RemoteHardware type=READ_GPIOS_REPLY, gpio_value=4 value=4

closed 2024-03-13 by Kralg

GUVWAF, 2024-03-15:
> Can you please write why the remoteHardwareModule is not added to modules vector the same as textMessageModule?

Hmm, I think it is, each class derived from `MeshModule` is added to it, because this happens in the constructor: https://github.com/meshtastic/firmware/blob/a085c3ddb334fb4977f68980ebd2ef79e625c861/src/mesh/MeshModule.cpp#L25

> I am thinking that also I might be able to write a custom parser of message in ProcessMessage TextMessageModule::handleReceived(const meshtastic_MeshPacket &mp) and handle custom way the GPIOs.

Yes, that might work. It would be dangerous to add this in the default firmware as you don't necessarily want anyone (accidentally) controlling your GPIOs.

> One more thing do you know how could I create a response from ProcessMessage TextMessageModule::handleReceived(const meshtastic_MeshPacket &mp) in case I would like to respond to parsed message?

You would need to create the `allocReply()` method, similar to the ReplyModule:
https://github.com/meshtastic/firmware/blob/a085c3ddb334fb4977f68980ebd2ef79e625c861/src/modules/ReplyModule.cpp#L8