meshtastic/python #515: 📢 Meshtastic Python needs a new platform owner / maintainer 📢

thebentern, 2024-03-19:
There are many outstanding issues with the CI/CD actions and other quality of life issues that need to be addressed. This project has been without a dedicated maintainer for a while now and is in danger of soon becoming deprecated. 

Please respond to this issue if you are interested in taking on this role or helping out in some specific way.

ianmcorvidae, 2024-03-21:
I thought I'd give myself a bit of time to think about it, but based on your response in discord as to what this would entail[^1], I think I would be willing to become a maintainer for this. I'd definitely want to put a call for other developers/maintainers and bit of a roadmap/to-do list into the README, like the [meshtastic/network-management-client](https://github.com/meshtastic/network-management-client) has, since I'm sure I won't have overwhelming amounts of time to devote to it or anything, but getting this back into some sort of shape is something I've already been working on, and I have some ideas for other things to do & improve as well.

Obviously I don't make any decisions but figured I'd at least throw my hat in the ring!

[^1]: "Mostly keeping protobufs in sync and generated, reviewing / merging PRs and issues, and kicking off releases"

thebentern, 2024-03-21:
@ianmcorvidae Sold!  😄 🎉

closed 2024-03-21 by thebentern

GUVWAF, 2024-03-21:
Great @ianmcorvidae!