meshtastic/python #518: Error loading Python lib - Ubuntu meshtastic release error

nikscave, 2024-03-20:
running Linux ubuntu 4.15.0-213-generic x86_64

followed instructions on ubuntu release page, error seen same as closed issue 447 (https://github.com/meshtastic/python/issues/447)

user@ubuntu:/home$ meshtastic
[16496] Error loading Python lib '/tmp/_MEIcP97hv/libpython3.9.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /tmp/_MEIcP97hv/libpython3.9.so.1.0)

If the app is standalone, what dependencies with python 3.9 exist and how can I fix this? any help much appreciated. thanks
Nik

garthvh, 2024-03-20:
That is a pretty old kernel, does a pip install work?

nikscave, 2024-03-20:
no, more issues with pip install, bleak dependencies missing. Just a mess.

Hadn't realised I was old an older kernel, I'll look to upgrade, thanks

nikscave, 2024-03-20:
still no dice. Definitely a python thing. What  I dont understand is the version showing following pip3 installation is

plex@ubuntu:~$ pip3 show meshtastic
Name: meshtastic
Version: 1.2.95
Summary: Python API & client shell for talking to Meshtastic devices
Home-page: https://github.com/meshtastic/Meshtastic-python
Author: Kevin Hester
Author-email: kevinh@geeksville.com
License: MIT
Location: /usr/local/lib/python3.6/dist-packages
Requires: dotmap, pexpect, protobuf, pygatt, pypubsub, pyqrcode, pyserial, pyyaml, tabulate, timeago
Required-by: 

which looks to be very old?

garthvh, 2024-03-21:
That is super old

ianmcorvidae, 2024-03-22:
It looks like 1.3 alpha 1 is the first version that dropped support for Python 3.6, which is likely why you have such an old version of this library installed (it looks like that's what version you're using, from the location shown in `pip3 show`). I don't really think this can be expected to work on such an old version of Python, unfortunately. Given your system is also throwing errors related to Python 3.9 simultaneously, there may be more issues with your python install as well.

nikscave, 2024-03-22:
thanks, so the installer is building version dependencies based on what's installed? not what it needs? anyway to override that before I look to re-build the box ;-)

ianmcorvidae, 2024-03-24:
Okay, I think I figured out what's up here. The standalone installer bundles python 3.9, but I think older ubuntu versions (including 20.04 that I'm on) don't have the right glibc for what it's building on more recent ubuntu. Separately, when using pip install, it's (understandably) only installing a version that's marked as compatible with the local python version (not the one in the installer), which in your case is 3.6, which is unsupported, so you're getting a very old version of the CLI that won't really work.

I'll try at some point to experiment a bit with getting something that works at least for 20.04, but the workaround other than updating/rebuilding your system is probably using a more recent python version, one way or another, whether that's by system packages, docker, or whatever else.

ianmcorvidae, 2024-12-22:
I'm going to close this for now. The standalone build works fairly well, at least on newer versions than ubuntu 20.04, which is now getting quite old. I would still like to improve our standalone builds, but I'd also like to clean up the issues list for now!

closed 2024-12-22 by ianmcorvidae