meshtastic/python #520: Standalone Installation 2.3.1 --nodes - Aborting due to: No module named timeago.locales.en

francisuk1989, 2024-03-21:
I am having issue with --nodes, It gives an error of

francis@hp-fury-g5 mesh$ ./meshtastic231 --host 192.168.2.208 --nodes
```
Connected to radio
Aborting due to: No module named 'timeago.locales.en'
```

installed
`Pygments-2.17.2 SecretStorage-3.3.3 astroid-3.1.0 autopep8-2.1.0 coverage-7.4.4 dill-0.3.8 docutils-0.20.1 importlib-metadata-7.1.0 iniconfig-2.0.0 isort-5.13.2 jaraco.classes-3.3.1 jeepney-0.8.0 keyring-24.3.1 markdown-3.6 markdown-it-py-3.0.0 mccabe-0.7.0 mdurl-0.1.2 more-itertools-10.2.0 nh3-0.2.15 pdoc3-0.10.0 pkginfo-1.10.0 pluggy-1.4.0 pycodestyle-2.11.1 pylint-3.1.0 pytap2-2.3.0 pytest-8.1.1 pytest-cov-4.1.0 readme-renderer-43.0 requests-toolbelt-1.0.0 rfc3986-2.0.0 rich-13.7.1 tomlkit-0.12.4 twine-5.0.0 webencodings-0.5.1 zipp-3.18.1`

garthvh, 2024-03-21:
Is this the Ubuntu standalone?

francisuk1989, 2024-03-21:
> Is this the Ubuntu standalone?

i have tried doing first with ubuntu 22.04.4 LTS and using the standalone
Also tried cloning the github but same issue.

francisuk1989, 2024-03-21:
I can confium that Debian 12 works fine, For some reason, both Ubuntu and Arch linux are having issue.

On Debian i just installed
`pip install meshtastic`
so not sure on the Ubuntu standalone

closed 2024-03-21 by francisuk1989