meshtastic/python #522: [Feature Request] sending telemetry from weather station

francisuk1989, 2024-03-21:
Is it possible we can send telemetry data from our own weather stations, e.g
`meshtastic  --host 192.168.5.250 --send-telemetry --temp '27' --humidity '77.0'  --pressure '1013.25' ` `
So in the app, it will show under the nodes e.g
27c 77.0% 1013.25 hPa

geeksville, 2024-04-03:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/weather-information-under-my-node-name/11548/2

francisuk1989, 2024-04-03:
> This issue has been mentioned on **Meshtastic**. There might be relevant details there:
> https://meshtastic.discourse.group/t/weather-information-under-my-node-name/11548/2

That was myself :) however i wasn't sure where to post this as i wasn't sure if mostly at aimed at https://github.com/meshtastic/protobufs or here.