meshtastic/python #531: Link to Documentation deployement is offline / broken

Jorropo, 2024-03-30:
Hey I don't know if this is the best place to report this but following links to the docs on the:
> [Documentation/API Reference](https://python.meshtastic.org/)

in [`README.md`](./README.md) and https://pypi.org/project/meshtastic/ yields:
[image]

thx.

Jorropo, 2024-03-30:
Someone in Discord mentioned:
> I think they got taken down just recently because they were really out of date, or something? Pending a rewrite? A code update? Don't remember sorry

Then this link should be removed, a stale link with `DEPLOYMENT_NOT_FOUND` looks very unintentional. :slightly_smiling_face:

ianmcorvidae, 2024-03-30:
I've removed the link for now, thanks for the heads-up!

closed 2024-03-30 by ianmcorvidae

VideoFX, 2024-08-02:
Where to find docs on the API?

commit 896eeff1 (2024-03-30): Update codecov link, remove current docs link, update standalone installation docs link

fixes #529, fixes #531