meshtastic/python #535: Unable to report nodes due to: 'latin-1' codec can't encode characters in position 0-198: ordinal not in range(256)

paulcart1, 2024-04-01:
When I try and query the nodes from the Python CLI I get errors due to characters used in the names (not my nodes but others in the mesh)

CLI Version: 2.3.3

Device: Heltec Tracker V3: V1.1
firmware_version: 2.3.2.63df972
device_state_version: 22

**$  meshtastic --nodes**

```
Connected to radio
Aborting due to: 'latin-1' codec can't encode characters in position 0-198: ordinal not in range(256)
```

**$ meshtastic --info**           
```
Connected to radio

Owner: Meshtastic cd94 (cd94)
My info: { "myNodeNum": 3663646100, "rebootCount": 12, "minAppVersion": 30200 }
Metadata: { "firmwareVersion": "2.3.2.63df972", "deviceStateVersion": 22, "canShutdown": true, "hasWifi": true, "hasBluetooth": true, "positionFlags": 811, "hwModel": "HELTEC_WIRELESS_TRACKER" }

Nodes in mesh: {"!da5ecd94": {"num": 3663646100, "user": {"id": "!da5ecd94", "longName": "Meshtastic cd94", "shortName": "cd94", "macaddr": "34:b7:da:5e:cd:94", "hwModel": "HELTEC_WIRELESS_TRACKER"}, "position": {"time": 1711994333}, "lastHeard": 1711994333, "deviceMetrics": {"batteryLevel": 32, "voltage": 3.556, "channelUtilization": 3.5149999, "airUtilTx": 0.020055557}}, "!abf88128": {"num": 2885189928, "user": {"id": "!abf88128", "longName": "Paul C", "shortName": "PC", "macaddr": "24:62:ab:f8:81:28", "hwModel": "TBEAM"}, "position": {"latitudeI": 512248750, "longitudeI": -11430632, "altitude": 196, "time": 1711968011, "latitude": 51.224875, "longitude": -1.1430632}, "snr": 6.0, "lastHeard": 1711968012, "deviceMetrics": {"batteryLevel": 83, "voltage": 3.999, "channelUtilization": 5.3783336, "airUtilTx": 0.4549722}}, "!da64ab8c": {"num": 3664030604, "user": {"id": "!da64ab8c", "longName": "roger_ab8c", "shortName": "rgr_", "macaddr": "34:b7:da:64:ab:8c", "hwModel": "HELTEC_V3"}, "snr": 6.5, "lastHeard": 1711904770, "deviceMetrics": {"batteryLevel": 53, "voltage": 3.746, "channelUtilization": 9.413333, "airUtilTx": 2.7508056}}, "!08bb76cc": {"num": 146503372, "user": {"id": "!08bb76cc", "longName": "M7OSA Wycombe", "shortName": "M7HW", "macaddr": "64:b7:08:bb:76:cc", "hwModel": "TLORA_V2_1_1P6", "isLicensed": true}, "snr": 6.5, "lastHeard": 1711967149, "deviceMetrics": {"batteryLevel": 101, "voltage": 4.312, "channelUtilization": 10.363334, "airUtilTx": 5.9311943}}, "!da5680dc": {"num": 3663102172, "user": {"id": "!da5680dc", "longName": "Beggarwood FS01", "shortName": "FS01", "macaddr": "34:b7:da:56:80:dc", "hwModel": "HELTEC_V3"}, "snr": 6.25, "lastHeard": 1711903619, "deviceMetrics": {"batteryLevel": 87, "voltage": 4.034, "channelUtilization": 12.821667, "airUtilTx": 2.90925}}, "!75f17f30": {"num": 1978761008, "user": {"id": "!75f17f30", "longName": "terahertz g McGill ", "shortName": "thzg", "macaddr": "dc:54:75:f1:7f:30", "hwModel": "TLORA_T3_S3"}, "snr": 6.5, "lastHeard": 1711904818}, "!b8241dd4": {"num": 3089374676, "user": {"id": "!b8241dd4", "longName": "Crowthorne NE RAK mobile", "shortName": "CRNm", "macaddr": "d9:ab:b8:24:1d:d4", "hwModel": "RAK4631"}, "snr": 6.5, "lastHeard": 1711904912}, "!da545a20": {"num": 3662961184, "user": {"id": "!da545a20", "longName": "Beggarwood FS02", "shortName": "FS02", "macaddr": "34:b7:da:54:5a:20", "hwModel": "HELTEC_V3"}, "snr": 6.25, "lastHeard": 1711967035, "deviceMetrics": {"batteryLevel": 91, "voltage": 4.07, "channelUtilization": 13.635, "airUtilTx": 2.2927501}}, "!da659688": {"num": 3664090760, "user": {"id": "!da659688", "longName": "MasterG1", "shortName": "MsG1", "macaddr": "34:b7:da:65:96:88", "hwModel": "HELTEC_V3"}, "snr": 6.0, "lastHeard": 1711967148}, "!da53f0e4": {"num": 3662934244, "user": {"id": "!da53f0e4", "longName": "Crowthorne NE 1", "shortName": "CRN1", "macaddr": "34:b7:da:53:f0:e4", "hwModel": "HELTEC_V3"}}}

Preferences: { "device": { "serialEnabled": true, "nodeInfoBroadcastSecs": 10800 }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsUpdateInterval": 120, "positionFlags": 811, "rxGpio": 33, "txGpio": 34, "broadcastSmartMinimumDistance": 100, "broadcastSmartMinimumIntervalSecs": 30, "gpsEnGpio": 37, "gpsMode": "ENABLED" }, "power": { "waitBluetoothSecs": 60, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_868", "hopLimit": 3, "txEnabled": true, "txPower": 27 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats", "encryptionEnabled": true, "root": "msh/EU_868" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900, "airQualityInterval": 900 }, "cannedMessage": {}, "audio": {}, "remoteHardware": {}, "neighborInfo": { "updateInterval": 900 }, "ambientLighting": { "current": 10, "red": 94, "green": 205, "blue": 148 }, "detectionSensor": { "minimumBroadcastSecs": 45, "detectionTriggeredHigh": true }, "paxcounter": {} }

Channels:
  Index 0: PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://meshtastic.org/e/#CgMSAQESCggBOANAA0gBUBs

```

ianmcorvidae, 2024-04-05:
I think that your system may be set up in a way that doesn't handle unicode correctly, so it's trying to use latin-1. It looks like the characters it's having trouble printing are the ones used to format the table, not the ones used for any node names.

If you've got a python setup and are willing, could you try running https://github.com/ianmcorvidae/meshtastic-python-scripts/blob/main/nodes.py ? (You should be able to download it, and then run it with `python3 nodes.py` or similar, as long as you have the `meshtastic` python module installed). That's a more barebones script that prints out the nodes without the table formatting, so if it works for you we can be confident it's the table formatting that's the issue, at least (though the nodes output within `--info` already suggests that, too).

paulcart1, 2024-04-08:
Thanks @ianmcorvidae  The barebones script worked and it turns out I had the wrong loacle set. Correctly setting locale to "en_GB.UTF-8" resolved the issue

closed 2024-04-08 by paulcart1