meshtastic/python #536: Add support for add channels query string to --seturl

garthvh, 2024-04-04:
Append any channels in a channels URL to the end of the array if the add=true query string value is present.

Otherwise delete and replace all channels (exiting functionality)

mikeymakesit, 2024-12-02:
I opened MR #708 which creates new option `--ch-add-url` that adds any new channels from an `add` URL.  It doesn't change any existing channels.  It errors out with relevant messages if the URL is invalid or the node runs out of channels.

I don't know pytest so I didn't contibute any tests for it in `test_node.py`.

Examples of use:

Add channels with new names from the URL:
```bash
% meshtastic --ch-add-url 'https://meshtastic.org/e/?add=true#Ch0SEAdk8ukuvs5lth8X1TQ5qlkaBFRlc3QlBAAAAAoeEhCklt7p_KHfBjTQmokk2D4FGgVUZXN0MiUFAAAAEg8IATgBQANIAVAeaAHABgE'
Connected to radio
Adding new channel 'Test' to device
Adding new channel 'Test2' to device
```

Fail when a non-add URL was provided:
```bash
% meshtastic --ch-add-url 'https://meshtastic.org/e/#Ch0SEAdk8ukuvs5lth8X1TQ5qlkaBFRlc3QlBAAAAAoeEhCklt7p_KHfBjTQmokk2D4FGgVUZXN0MiUFAAAAEg8IATgBQANIAVAeaAHABgE'
Connected to radio
Warning: Invalid URL 'https://meshtastic.org/e/#Ch0SEAdk8ukuvs5lth8X1TQ5qlkaBFRlc3QlBAAAAAoeEhCklt7p_KHfBjTQmokk2D4FGgVUZXN0MiUFAAAAEg8IATgBQANIAVAeaAHABgE'
```

mikeymakesit, 2024-12-02:
I chose a new argument instead of overloading `--seturl` because it feels like a different behavior.  But I can refactor it into `--seturl` if you want to reduce new options

juergenRe, 2025-10-30:
I don't have the permissions to do so... @garthvh : can you please close it?