meshtastic/python #537: BUG: The Python client is not able to change the value of position.fixed_position over serial

hawke114, 2024-04-06:
The Python library is unable to set the value of position.fixed_position.

The Python client will accept the command to set the position.fixed_position value via serial, but the value does not change unless set in the web client or Android app.

Details of my setup below, happy to provide additional details.

#### Set and Get commands

```
>meshtastic --set position.fixed_position false
Connected to radio
Set position.fixed_position to false
Writing modified preferences to device
```

```
>meshtastic --get position.fixed_position
Connected to radio
position.fixed_position: True
Completed getting preferences
```
#### Client version:
```
meshtastic --version
2.3.3
```
#### OS: Windows 10
#### Device metadata:
```
{
    "firmwareVersion": "2.3.2.63df972",
    "deviceStateVersion": 22,
    "canShutdown": true,
    "hasWifi": true,
    "hasBluetooth": true,
    "positionFlags": 811,
    "hwModel": "HELTEC_V3"
}
```

#### Device preferences:
```
{
    "device": {
        "serialEnabled": true,
        "nodeInfoBroadcastSecs": 10800
    },
    "position": {
        "positionBroadcastSecs": 900,
        "positionBroadcastSmartEnabled": true,
        "fixedPosition": true,
        "gpsUpdateInterval": 120,
        "positionFlags": 811,
        "broadcastSmartMinimumDistance": 100,
        "broadcastSmartMinimumIntervalSecs": 30,
        "gpsMode": "NOT_PRESENT"
    },
    "power": {
        "waitBluetoothSecs": 60,
        "sdsSecs": 4294967295,
        "lsSecs": 300,
        "minWakeSecs": 10
    },
    "network": {
        "ntpServer": "0.pool.ntp.org"
    },
    "display": {
        "screenOnSecs": 600
    },
    "lora": {
        "usePreset": true,
        "hopLimit": 3,
        "txEnabled": true,
        "txPower": 30,
        "sx126xRxBoostedGain": true
    },
    "bluetooth": {
        "enabled": true,
        "fixedPin": 123456
    }
}
```
#### Module Preferences (default)
```
{
    "mqtt": {
        "address": "mqtt.meshtastic.org",
        "username": "meshdev",
        "password": "large4cats",
        "encryptionEnabled": true,
        "root": "msh/US"
    },
    "serial": {},
    "externalNotification": {},
    "rangeTest": {},
    "telemetry": {
        "deviceUpdateInterval": 900,
        "environmentUpdateInterval": 900,
        "airQualityInterval": 900
    },
    "cannedMessage": {},
    "audio": {},
    "remoteHardware": {},
    "neighborInfo": {
        "updateInterval": 900
    },
    "ambientLighting": {
        "current": 10,
        "red": 91,
        "green": 6,
        "blue": 40
    },
    "detectionSensor": {
        "minimumBroadcastSecs": 45,
        "detectionTriggeredHigh": true
    },
    "paxcounter": {}
}
```

hawke114, 2024-04-07:
Tested with python cli over wifi and it is successful, so this seems to be limited to an issue with the serial interface to the radio.

CR-DMcDonald, 2024-05-07:
I am having the same issue using --setalt