meshtastic/python #550: Connecting --ble crashed the python

slash-bit, 2024-04-21:
Hi, I get the following error when connecting via BLE to any of my nodes. Tried with name or address of the device .

PS C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts> meshtastic --ble-scan
Found: name='0000_d46d' address='F9:D9:14:60:D4:6D'
BLE scan finished

PS C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts> meshtastic --ble 0000_d46d --noproto
INFO file:client.py pair line:618 Paired to device with protection level <DevicePairingProtectionLevel.NONE: 1>.
WARNING file:mesh_interface.py _sendToRadio line:729 Not sending packet because protocol use is disabled by noProto
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\__main__.py", line 1475, in main
    common()
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\__main__.py", line 1011, in common
    client = BLEInterface(args.ble, debugOut=logfile, noProto=args.noproto)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 70, in __init__
    self.client.start_notify(FROMNUM_UUID, self.from_num_handler)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 207, in start_notify
    self.async_await(self.bleak_client.start_notify(*args, **kwargs))
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 220, in async_await
    return self.async_run(coro).result(timeout)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\concurrent\futures\_base.py", line 458, in result
    return self.__get_result()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\concurrent\futures\_base.py", line 403, in __get_result
    raise self._exception
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\__init__.py", line 834, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 974, in start_notify
    await winrt_char.write_client_characteristic_configuration_descriptor_async(
OSError: [WinError -2140864507] The attribute requires authentication before it can be read or written

mkgin, 2024-06-10:
> File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 974, in start_notify
await winrt_char.write_client_characteristic_configuration_descriptor_async(
OSError: [WinError -2140864507] The attribute requires authentication before it can be read or written

The same with linux...  when not paired. It was discussed in Meshtastic # python channel in discord yesterday. I'll look at this in a few hours. Some exception handling maybe?

slash-bit, 2024-06-10:
Probable because its not paired. But how does one get paired?

I literally do scan
>meshtastic --ble-scan
Found: name='BIST_d46d' address='F9:D9:14:60:D4:6D'
BLE scan finished

And then connect. Never get prompted to pair.

>meshtastic --ble BIST_d46d
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\__main__.py", line 1560, in main
    common()
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\__main__.py", line 1053, in common
    client = BLEInterface(args.ble, debugOut=logfile, noProto=args.noproto, noNodes=args.no_nodes)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 64, in __init__
    self._startConfig()
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\mesh_interface.py", line 723, in _startConfig
    self._sendToRadio(startConfig)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\mesh_interface.py", line 754, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 155, in _sendToRadioImpl
    self.client.write_gatt_char(TORADIO_UUID, b, response = True)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 206, in write_gatt_char
    self.async_await(self.bleak_client.write_gatt_char(*args, **kwargs))
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\meshtastic\ble_interface.py", line 222, in async_await
    return self.async_run(coro).result(timeout)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\concurrent\futures\_base.py", line 458, in result
    return self.__get_result()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\concurrent\futures\_base.py", line 403, in __get_result
    raise self._exception
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\__init__.py", line 776, in write_gatt_char
    await self._backend.write_gatt_char(characteristic, data, response)
  File "C:\Users\v-b2\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 887, in write_gatt_char
    raise BleakError("Not connected")
bleak.exc.BleakError: Not connected

mkgin, 2024-06-10:
Meshtastic python-cli doesn't do it yet... but the may be a prompt that shows elsewhere regarding the pin. not sure for windows...  but in linux shell it's bluetoothctl .. depending on how things are set up there may be a notification or popup in a gui or nothing.

...

slash-bit, 2024-06-11:
No, noting pops up else where on a windows.
Jus looking a the code, there is something about client.pair in bleinterface.py. The exception seems to be related to "NotImplementederror" only.
I haven't looked properly. I am looking at something else  ( figuring out  how to sent traceroute within python)

I will check later.
[image]

    def connect(self, address):
        "Connect to a device by address"
        device = self.find_device(address)
        client = BLEClient(device.address)
        client.connect()
        try:
            client.pair()
        except NotImplementedError:
            # Some bluetooth backends do not require explicit pairing.
            # See Bleak docs for details on this.
            pass
        return client

slash-bit, 2024-06-11:
> The same with linux... when not paired. It was discussed in Meshtastic # python channel in discord yesterday. I'll look at this in a few hours. Some exception handling maybe?

Hi Could you, please send me link to that Discord channel. I seems don't have it.? Thanks