meshtastic/python #565: Failing tests on Alpine Linux

OTLabs, 2024-05-05:
I am building the meshtastic package for Alpine Linux and I got 5 tests failing:

```
=========================== short test summary info ============================
FAILED meshtastic/tests/test_mesh_interface.py::test_MeshInterface - TypeError: MessageToJson() got an unexpected keyword argument 'always_print...
FAILED meshtastic/tests/test_node.py::test_node - TypeError: MessageToJson() got an unexpected keyword argument 'always_print...
FAILED meshtastic/tests/test_serial_interface.py::test_SerialInterface_single_port - TypeError: MessageToJson() got an unexpected keyword argument 'always_print...
FAILED meshtastic/tests/test_tcp_interface.py::test_TCPInterface - TypeError: MessageToJson() got an unexpected keyword argument 'always_print...
FAILED meshtastic/tests/test_util.py::test_message_to_json_shows_all - TypeError: MessageToJson() got an unexpected keyword argument 'always_print...
================ 5 failed, 196 passed, 75 deselected in 16.71s =================

```

I would quite appreciate any comments and invites about the possible origin of there errors and how to solve them.

ianmcorvidae, 2024-05-05:
Most likely scenario is that your version of the protobuf library is too old; that argument was added somewhat recently in that library. Make sure you have protobuf>=5.26.0

OTLabs, 2024-05-05:
I got py3-protobuf (4.24.4-r1), let me see how to upgrade it. Thanks a lot!

ianmcorvidae, 2024-05-09:
I'm going to close this for now, but feel free to comment if upgrading doesn't fix things. Pretty likely that's the issue, though!

closed 2024-05-09 by ianmcorvidae