meshtastic/python #571: Raspberry Pi Python CLI unexpected behaviour - node crash

joneyes, 2024-05-10:
Prerequisites:
Pi 3B+, Bullseye. Python v3.9.2
Meshtastic v2.3.8
Lilygo T3S3, running 2.3.7b

Issuing a --request-telemetry command from the pi terminal results in unexpected behaviour, which can crash the node:
It appears that the destination field is being replaced, this can be seen in the echo'd command:
_pi@pimesh:~ $ meshtastic --dest !04ba88d0 --request-telemetry
meshtastic --dest **ifconfigba88d0** --request-telemetry
Connected to radio
Sending telemetry request to ifconfigba88d0 on channelIndex:0 (this could take a while)
Warning: NodeId ifconfigba88d0 not found in DB_

Issuing the command via a --host address from the Pi terminal has the same effect:

_pi@pimesh:~ $ meshtastic --host 192.168.1.251 --dest !04ba88d0 --request-telemetry
meshtastic --host 192.168.1.251 --dest ifconfigba88d0 --request-telemetry
Connected to radio
Sending telemetry request to ifconfigba88d0 on channelIndex:0 (this could take a while)
Warning: NodeId ifconfigba88d0 not found in DB_

Issuing the same command from a Windows Shell is correctly interpreted:

_PS D:\meshtastic> .\meshtastic --host 192.168.1.251 --dest !04ba88d0 --request-telemetry
Connected to radio
Sending telemetry request to !04ba88d0 on channelIndex:0 (this could take a while)
Telemetry received:
Battery level: 99.00%
Voltage: 4.19 V
Total channel utilization: 4.91%
Transmit air utilization: 0.35%
PS D:\meshtastic>_

ianmcorvidae, 2024-05-10:
You need to put single quotes around the `dest` option. That `ifconfig` in there is due to the shell expanding out the `!` into a previous command (specifically, your 4th-most recent, I believe, from the `04`. In other words, it's not meshtastic -- it's the system itself.

You should see what you want with `meshtastic --dest '!04ba88d0' --request-telemetry` instead

closed 2024-05-10 by ianmcorvidae

joneyes, 2024-05-11:
Doh. Linux idiosyncrasies.
Tnx Ian

Regards,
Jon Eyes