meshtastic/python #573: [Bug]: Using yaml file to configure "modemPreset" and "region" causes the settings not to be applied.

Wireheadbe, 2024-05-13:
using ``` meshtastic --host xxx.xxx.xxx.xxx --configure file.yaml```

With the following lora config:
```
config:
  bluetooth:
    enabled: true
    fixedPin: 123456
  device:
    nodeInfoBroadcastSecs: 10800
    role: CLIENT
    serialEnabled: true
  display:
    screenOnSecs: 600
  lora:
    hopLimit: 20
    modemPreset: LONG_SLOW
    region: EU_433
    txPower: 20
```
.. the "modemPreset" and "region" do not get applied on a freshly "wiped and flashed" device.

Hardware:
Heltec v3.1

Firmware:
2.3.7.30fbcab

Meshtastic Python API:
2.3.8

Wireheadbe, 2024-05-13:
Doing it manually:
 meshtastic --host 192.168.201.42 --set lora.region EU_433
Connected to radio
Set lora.region to EU_433
Writing modified preferences to device

-> Works, But the device didn't reboot immediately. Only after executing it a second time. 
-> After reboot, the region is set.
-> Not sure if this is purely a python issue, or firmware issue

Wireheadbe, 2024-05-13:
Same thing for modem_preset - via config file: not applied. Trying manually seems to work.
```
Connected to radio
DEBUG file:mesh_interface.py sendPosition line:384 p.time:1715626829
DEBUG file:mesh_interface.py sendData line:318 Serializing protobuf as data: time: 1715626829
DEBUG file:mesh_interface.py sendData line:321 len(data): 5
DEBUG file:mesh_interface.py sendData line:322 mesh_pb2.Constants.DATA_PAYLOAD_LEN: 237
DEBUG file:mesh_interface.py _sendPacket line:583 Sending packet: to: 4294967295 decoded { portnum: POSITION_APP payload: "%McBf" } id: 3301257290 hop_limit: 3
DEBUG file:stream_interface.py _sendToRadioImpl line:109 Sending: packet { to: 4294967295 decoded { portnum: POSITION_APP payload: "%McBf" } id: 3301257290 hop_limit: 3 }
DEBUG file:stream_interface.py _sendToRadioImpl line:114 sending header:b'\x94\xc3\x00\x19' b:b'\n\x17\x15\xff\xff\xff\xff"\t\x08\x03\x12\x05%McBf5J0\xc5\xc4H\x03'
DEBUG file:mesh_interface.py _sendToRadio line:777 packet c4c5304a got acked under us
DEBUG file:__main__.py setPref line:163 snake_name:modem_preset
DEBUG file:__main__.py setPref line:164 camel_name:modemPreset
DEBUG file:__main__.py setPref line:185 valStr:LONG_SLOW val:LONG_SLOW
Set lora.modem_preset to LONG_SLOW
Writing modified preferences to device
DEBUG file:node.py writeConfig line:208 Wrote: lora
DEBUG file:node.py _sendAdmin line:794 adminIndex:0
DEBUG file:mesh_interface.py sendData line:318 Serializing protobuf as data: set_config { lora { use_preset: true modem_preset: LONG_SLOW region: EU_433 hop_limit: 3 tx_enabled: true tx_power: 12 sx126x_rx_boosted_gain: true ignore_mqtt: true } }
DEBUG file:mesh_interface.py sendData line:321 len(data): 22
DEBUG file:mesh_interface.py sendData line:322 mesh_pb2.Constants.DATA_PAYLOAD_LEN: 237
DEBUG file:mesh_interface.py _sendPacket line:583 Sending packet: to: 3662950296 decoded { portnum: ADMIN_APP payload: "\222\002\0232\021\010\001\020\0018\002@\003H\001P\014h\001\300\006\001" want_response: true } id: 3301257291 hop_limit: 3
DEBUG file:stream_interface.py _sendToRadioImpl line:109 Sending: packet { to: 3662950296 decoded { portnum: ADMIN_APP payload: "\222\002\0232\021\010\001\020\0018\002@\003H\001P\014h\001\300\006\001" want_response: true } id: 3301257291 hop_limit: 3 }
DEBUG file:stream_interface.py _sendToRadioImpl line:109 Sending: heartbeat { }
DEBUG file:stream_interface.py _sendToRadioImpl line:114 sending header:b'\x94\xc3\x00,' b:b'\n*\x15\x98/T\xda"\x1c\x08\x06\x12\x16\x92\x02\x132\x11\x08\x01\x10\x018\x02@\x03H\x01P\x0ch\x01\xc0\x06\x01\x18\x015K0\xc5\xc4H\x03'
DEBUG file:stream_interface.py _sendToRadioImpl line:114 sending header:b'\x94\xc3\x00\x02' b:b':\x00'
DEBUG file:mesh_interface.py _sendToRadio line:777 packet c4c5304b got acked under us
DEBUG file:tcp_interface.py close line:58 Closing TCP stream
DEBUG file:stream_interface.py close line:119 Closing stream
DEBUG file:stream_interface.py _sendToRadioImpl line:109 Sending: disconnect: true
DEBUG file:stream_interface.py _sendToRadioImpl line:114 sending header:b'\x94\xc3\x00\x02' b:b' \x01'
DEBUG file:stream_interface.py __reader line:202 reader is exiting
DEBUG file:stream_interface.py _disconnected line:80 Closing our port
```

Wireheadbe, 2024-05-13:
Same issue for hopLimit, txPower,..

Also one observation, is that whenever these settings are applied: the reboot afterwards is hit-and-miss. 
I'm not sure if this is purely related to the Python API, or the firmware. I've seen e.g. https://github.com/meshtastic/python/issues/41 - also https://meshtastic.discourse.group/t/preferences-settings-sometimes-not-sticking-saving-cli-2-2-22-firmware-2-2-24-app-2-2-24/10560/2 seems relevant.

thebentern, 2024-05-14:
So it would seem it isn't rebooting reliably after the configure imports the yaml?

Wireheadbe, 2024-05-14:
Correct - tested on 2 devices, with full erase + firmware load. 
But, when it _does_ reboot - it doesn't update the settings.

geoffreygarrett, 2024-05-14:
> #576 seems related.

I've since found a solution on my end for my requirements. **Something worth noting** is that *if* you are setting the `channel_url` in your config (as I was) it too has a `lora_config` protobuf contained within. Of the two observations I made in #576, this seems to be what caused the second. The first problem, related to the `--host` TCP interface is probably what you're dealing with (if your `yaml` is complete above, and didn't have the `channel_url` omitted for brevity).

vogelh0ws, 2024-12-19:
I experience this problem too.

Does someone have an explanation why this is open since May and not even confirmed?

From my understanding exporting and restoring configs is quite a common task.

zero2005x, 2026-09-27:
Seeing what looks like the same problem on firmware 2.8.0.47db0e3 with CLI 2.7.11 (Windows 11, USB serial), on 2× Heltec V4 TFT and 1× Heltec V3, each right after a full erase + flash:

- `meshtastic --configure backup.yaml` (exported from the same node on 2.7.14: region, owner, `channel_url`, bluetooth, security keys, modules) reports success, but after the reboot the node is back to defaults. The boot log shows:

  ```
  ERROR | Can't open/read /prefs/config.proto
  INFO  | Install default LocalConfig
  INFO  | Wanted region 0, using UNSET
  ```

  On one node a new key pair (and node number) was generated on every boot until the config was restored.
- Workaround, 3 of 3 nodes: `--set lora.region <REGION>` on its own first, then the same `--configure` succeeds (`Loaded /prefs/config.proto`, `Wanted region 8, using TW`).
- On a node that already has its key pair I could *not* reproduce it: setting the region back to UNSET and running `--configure backup.yaml --wait-to-disconnect 20` saved everything. So the trigger seems to be the freshly erased state (no key pair yet, `Skip NodeDB without key` in the boot log), where the first region set generates the keys and switches the node number while the configure transaction is in flight. That would fit the "freshly wiped and flashed" report above.

Caveat: on the two V4 nodes (ESP32-S3 USB-Serial/JTAG), each CLI connect/disconnect also reboots the node on Windows, which muddies those first attempts. The V3 run is the cleanest evidence: there I opened the port with DTR/RTS deasserted and waited 20 s before closing, so the node was not reset during the transaction. Possibly related: #850.

_Investigated with the help of Claude Code; results reproduced on real hardware._