meshtastic/python #575: [Bug]:  export -import  function not include some parameter

bv3un, 2024-05-14:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.9.f06c56a

### Description

Export and import function not include some parameter like GPS position ,NTP server , licensed amateur radio.thus after firmware upgrade and import config need to setup again

ianmcorvidae, 2025-01-09:
NTP server is exported by `--export-config`, as is fixed position. I'm not sure where you're seeing it not export position, though it may not do so if you aren't using fixed position (since it wouldn't be useful to do so anyway). It does appear that the `is_licensed` field isn't exported along with the other ownership fields, though. Position precision only needs to be exported in a couple places. One is in the mqtt settings under mapReportSettings, where it is; the other is in channel settings, which is included within the channel URL. Within `--info`, position and altitude will be within the `nodes` data structure -- they're not configuration parameters in the same way other things are.

Other than the is_licensed field I don't think there's anything here to fix.

bplein, 2025-01-09:
Thank you for responding so quickly!

For `--export-config`, `fixedPosition: <boolean>` is exported, but not the actual position information. This means that setting a fixed position on a node is not backed up, and replacing or re-imaging that node, the full configuration cannot be restored.

Example from my export
```
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 60
    fixedPosition: true
    gpsUpdateInterval: 120
    positionBroadcastSecs: 10800
    positionBroadcastSmartEnabled: true
    positionFlags: 3
```

I found this by having to restore a setting to a node, where i have a fixed position that is not exactly where the node is located. I checked from another node, and it was reporting the position of my phone, not the expected values from my previous `--setlat` etc.

ianmcorvidae, 2025-01-09:
The position information is exported under the `location` header, outside of the `config` block, for reasons that have to do with the firmware implementation. The `position` settings section controls _how_ position is managed, vs what actual location is used, basically.

In your export you should see first a `channel_url`, then `config' with a number of sub-sections, then `location`, `module_config` with subsections, and finally `owner` and `owner_short`

bplein, 2025-01-09:
Interesting. My T-beam is correctly showing `location` header by my RAK is not.  I was only looking at my RAK4631 device when I commented on this issue.

I wonder if there's another setting that might be preventing `location` header from being populated.

Thank you for your patience with me. It's on my end it appears.

interfect, 2025-05-11:
`--configure` dropping large swathes of the config is I think caused by the CLI tool not waiting for the earlier transaction/config packets to be handled by the node before sending the later ones, which makes the node drop earlier packets. See https://github.com/meshtastic/firmware/issues/5640#issuecomment-2869179446

I think `_sendAdmin` either needs to wait for each admin packet to actually be acknowledged (or time out?) before sending the next one:
https://github.com/meshtastic/python/blob/622a435465872fd316054f1a4a7dd6d4d44bb3ec/meshtastic/node.py#L991-L1000

Or else we need to add a wall-clock delay between admin packets when restoring the config.