meshtastic/python #576: [BUG]: --configure not effective with .yaml and some options

geoffreygarrett, 2024-05-14:
## Versions and Firmware 

```
(base) ➜  firmware-2 meshtastic --version
2.3.8

(base) ➜  firmware-2 meshtastic --info
My info: { "myNodeNum": 3771639288, "rebootCount": 3, "minAppVersion": 30200 }
Metadata: { "firmwareVersion": "2.3.8.d490a33", "deviceStateVersion": 22, "canShutdown": true, "hasWifi": true, "hasBluetooth": true, "positionFlags": 811, "hwModel": "TBEAM", "hasEthernet": false, "role": "CLIENT", "hasRemoteHardware": false }
```

## `--set` has no effect with `--host`

```
(base) ➜  nsri-meshtastic meshtastic --set lora.ignore_mqtt false --host 192.168.68.128
Connected to radio
Set lora.ignore_mqtt to false
Writing modified preferences to device

(base) ➜  nsri-meshtastic meshtastic --export-config --host 192.168.68.128             
# start of Meshtastic configure yaml
channel_url: ...
config:
  ...
  lora:
   ignoreMqtt: true

...
```

## `--configure` doesn't change `ignoreMqtt`...

### `e0cea5f8.yaml`

```yaml
# Meshtastic Configuration YAML
channel_url: '{{ _.channel_url }}'
owner: '{{ long_name }}'
owner_short: '{{ short_name }}'

config:
  bluetooth:
    enabled: true
    mode: RANDOM_PIN
  lora:
    region: EU_868
{#    ignore_mqtt: false#}
  network:
    wifi_enabled: false
    wifi_ssid: '{{ _.network.wifi_ssid }}'
    wifi_psk: '{{ _.network.wifi_psk }}'
    address_mode: DHCP
    eth_enabled: false

module_config:
  mqtt:
    enabled: true
    encryption_enabled: true
    map_report_settings:
      position_precision: 15
      publish_interval_secs: {{ _.mqtt.publish_interval_secs }}
    map_reporting_enabled: true
    proxy_to_client_enabled: false
    root: '{{ _.mqtt.root }}'
    address: '{{ _.mqtt.address }}'
    username: '{{ _.mqtt.username }}'
    password: '{{ _.mqtt.password }}'
    tls_enabled: true
  telemetry:
    air_quality_enabled: true
    air_quality_interval: 900
    device_update_interval: 900
    environment_display_fahrenheit: false
    environment_measurement_enabled: true
    environment_screen_enabled: true
    environment_update_interval: 900
    power_measurement_enabled: false
    power_screen_enabled: false
    power_update_interval: 0
```

It doesn't matter if I have it commented out or not, I simply cannot configure a from scratch, without the UI due to the `ignoreMqtt` not being affected by any CLI commands. I have tried both snake_case and camelCase, but I can see internally it's converting regardless.

```
(base) ➜  nsri-meshtastic meshtastic --factory-reset && meshtastic --reset-nodedb && meshtastic --configure config/e0cea5f8.yaml

Setting device owner to [Device Owner]
Setting device owner short to [Owner Short]
Setting channel url to [URL]
Set bluetooth.enabled to True
Set bluetooth.mode to RANDOM_PIN
Set lora.region to EU_868
Set network.wifi_enabled to False
Set network.wifi_ssid to [SSID]
Set network.wifi_psk to [PSK]
Set network.address_mode to DHCP
Set network.eth_enabled to False
Set mqtt.enabled to True
Set mqtt.encryption_enabled to True
Set mqtt.map_report_settings.position_precision to 15
Set mqtt.map_report_settings.publish_interval_secs to 0
Set mqtt.map_reporting_enabled to True
Set mqtt.proxy_to_client_enabled to False
Set mqtt.root to msh/ZA
Set mqtt.address to [MQTT Address]
Set mqtt.username to [Username]
Set mqtt.password to [Password]
Set mqtt.tls_enabled to True
Set telemetry.air_quality_enabled to True
Set telemetry.air_quality_interval to 900
Set telemetry.device_update_interval to 900
Set telemetry.environment_display_fahrenheit to False
Set telemetry.environment_measurement_enabled to True
Set telemetry.environment_screen_enabled to True
Set telemetry.environment_update_interval to 900
Set telemetry.power_measurement_enabled to False
Set telemetry.power_screen_enabled to False
Set telemetry.power_update_interval to 0
```

```
meshtastic --export-config > config/resultant_e0cea5f8.yaml  
```

```yaml
# start of Meshtastic configure yaml
channel_url: '[Channel URL]'
config:
  bluetooth:
    enabled: true
    fixedPin: [Fixed Pin]
  device:
    nodeInfoBroadcastSecs: 10800
    serialEnabled: true
  display:
    screenOnSecs: 600
  lora:
    hopLimit: 3
    ignoreMqtt: true
    region: EU_868
    sx126xRxBoostedGain: true
    txEnabled: true
    txPower: 27
    usePreset: true
  network:
    ntpServer: 0.pool.ntp.org
    wifiPsk: '[WiFi PSK]'
    wifiSsid: '[WiFi SSID]'
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    gpsMode: ENABLED
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionBroadcastSmartEnabled: true
    positionFlags: 811
  power:
    lsSecs: 300
    minWakeSecs: 10
    sdsSecs: 4294967295
    waitBluetoothSecs: 60
module_config:
  ambientLighting:
    blue: 248
    current: 10
    green: 165
    red: 206
  detectionSensor:
    detectionTriggeredHigh: true
    minimumBroadcastSecs: 45
  mqtt:
    address: '[MQTT Address]'
    enabled: true
    encryptionEnabled: true
    mapReportSettings:
      positionPrecision: 15
    mapReportingEnabled: true
    password: '[MQTT Password]'
    root: 'msh/ZA'
    tlsEnabled: true
    username: '[MQTT Username]'
  neighborInfo:
    updateInterval: 900
  paxcounter:
    paxcounterUpdateInterval: 120
  rangeTest:
    sender: 15
  telemetry:
    airQualityEnabled: true
    airQualityInterval: 900
    deviceUpdateInterval: 900
    environmentMeasurementEnabled: true
    environmentScreenEnabled: true
    environmentUpdateInterval: 900
owner: '[Owner]'
owner_short: '[Owner Short]'
```

This is a bit of an issue for me, I need a simple streamline way of configuring devices from scratch, and then updating via CI/CD at a later time remotely. I was hoping `--configure` would be effective but it seems not.

yaleman, 2024-05-27:
I'm having similar issues trying to use the meshtastic client and configuring mqtt settings.

It'll routinely just not update the root, or ... most other settings on change. Try it a few times and it'll come good, sometimes.

ianmcorvidae, 2024-06-08:
I have a feeling that there's possibly some more we could be doing in the CLI to send these commands better or to better verify they've gone through -- leaving a note here to remind myself but also in case anyone else feels up to it, but probably comparing with what the Android and iOS apps (especially Android since it supports TCP) are doing would be the place to start on this. It's also possible that adding the `--wait-to-disconnect` argument will improve the behavior in some of these regards, but that's all a little unclear to me at present.

vogelh0ws, 2024-12-19:
I experience this problem too.

Does someone have an explanation why this is open since May and not even confirmed?

From my understanding exporting and restoring configs is quite a common task.

garthvh, 2024-12-19:
I don't hear from many users using it.

todd2982, 2025-01-15:
Is this due to #728 ?  I've found that the export writes the file with the wrong type of encoding, and the configure command will only import UTF8 files.

geoffreygarrett, 2025-01-15:
Maybe something worthwhile keeping an eye on, and providing feedback on, is a new compiled Meshtastic CLI in Rust, which is aimed at being more reliable and cross-platform. The initial discussions are here, with two of us having made progress. We will be looking for feedback on the `v2` design, but `v1` will still provide the same experience as the existing for easy migration.

- https://github.com/meshtastic/rust/issues/25

D0han, 2025-03-01:
> I don't hear from many users using it.

I do use that.
I am changing firmware quite often, so when i found export/import in cli i expected to that to be solid and to work correctly. It is not, so uninterrupted flow of - get new FW build, export, flash, import, test is not possible at the moment. Every time i need to pair with phone and set everything by hand as it was before..

dlitz, 2025-04-08:
I just ran into this and it's very frustrating and time-wasting.  The tool shouldn't just silently return success if configuration fails for some reason.

berlincount, 2025-04-22:
hmmm, I'm running into the same issue here. very confusing.

interfect, 2025-05-11:
This is I think partially a duplicate of #575.

interfect, 2025-08-21:
This should have been fixed by #783.