meshtastic/python #594: [Feature Request]: Confirmation of sent message

brad28b, 2024-06-13:
When using: 

> meshtastic --sendtext --dest xxxxx Text

or

> meshtastic -ch-index 0  --sendtext Text

It would be very nice to have a confirmation of the message being recieved. When sending to a channel, a confirmation that at least one node received the message. When send direct to a node, either confirmation that the particular node got the message, or at least, confirmation that least one other node heard the message. In the smartphone app, similar to the little tick that appears when a node gets your message, but in the case outputting in a parsable CLI format.

Current behaviour is it sends the message, and then exits. And, for whatever reason, the message is often not received, even between two close by nodes. Confirmation that it wasn't received will allow us to deal with that programatically.

GUVWAF, 2024-06-13:
You can add `--ack` to wait for an acknowledgement.

Which devices do you have? If the nodes are too close, they might get overloaded, especially if they have high transmit power. Put them at least a room away and you should get near 100% success rate.

ianmcorvidae, 2024-06-13:
In looking at this bug I did discover that `--ack` seems to be broken, and it'll also return with only an implicit ack, so there's definitely some improvement to be made. #570 also relevant here, something I'll definitely keep in mind there!

GUVWAF, 2024-06-13:
Oh, sorry, didn't know that. Good to hear you're on it.

rj45jack, 2024-06-13:
I am also having this issue where everything seems to have transmitted just fine in the console, but nothing goes out to the radio or seen by other nodes.

Phlat, 2024-07-10:
I was curious about that even with  my own experiments with the api and trying to setup some simple bot scripts to check the weather by zip code. Sometimes I have to send the command a couple times and can see it coming thru but no response.

rj45jack, 2024-07-10:
> I was curious about that even with my own experiments with the api and trying to setup some simple bot scripts to check the weather by zip code. Sometimes I have to send the command a couple times and can see it coming thru but no response.

My issue is the exact same use-case. I am trying to publish weather alerts, and general weather but it's hit or miss with TCP.

Phlat, 2024-07-10:
not BBS related. This is what I came up with, my weather lookup.
It's not direct message. For what it's worth. Cheers!

Apfelwurm, 2024-07-19:
We managed to finnaly buiild something that seems to handle ack/nak successfully: https://github.com/Apfelwurm/alertmanagermeshtastic/blob/main/src/alertmanagermeshtastic/meshtastic.py#L144-L211 for those interested :)