meshtastic/python #596: Enhancement: --request-environmental

joneyes, 2024-06-15:
It is possible to request the following via CLI for a remote node:
--request-position (returns position co-ords & alt)
--request-telemetry (returns current device metrics)

Is it possible to add --request-environmental to query environmental metric data, when enabled?

joneyes, 2024-06-16:
OK - I'll raise and enhancement request

joneyes, 2024-06-23:
Firmware to support this has now been updated.

ianmcorvidae, 2024-06-26:
Excellent, thanks for shepherding that. I'll try to get it implemented over here sometime soon, hopefully.

joneyes, 2024-10-12:
Did you make any progress on this? - specifically looking for the power metrics telemetry

ianmcorvidae, 2024-10-14:
I totally lost track of it, I'm sorry. I put together a PR to make this possible, which should go into the next release. I made it work by passing a second argument along with `--request-telemetry` to specify the subtype desired, rather than individual arguments for each type.

PR #687 (2024-10-14): Support requesting different telemetry types
This allows passing an argument along with `--request-telemetry`. The default, without passing an argument, will be the current behavior, requesting device telemetry, also possible with `--request-telemetry device`. Other types can be requested with relevant arguments (`environment`, `air_quality` or `airquality`, or `power`)

Fixes #596
closed 2024-10-14 by ianmcorvidae