meshtastic/python #598: {BUG] Emoji character in node name screws up tabular formatting of --nodes

brad28b, 2024-06-17:
See attached image. The node 'Highton Rooftop' is using an Emoji in their node name. It screws up the tabular formatting of the output for the --nodes argument, making it harder to parse, and visually messy.
[image]

ianmcorvidae, 2024-06-17:
This is a weakness of the underlying library, `tabulate`, but can be corrected by installing the library `wcwidth` (which is not included by default with `tabulate`.

If you run `pip install wcwidth` within the same environment you installed `meshtastic`, it should be corrected

brad28b, 2024-06-17:
Thank you very much the very fast response. I can confirm your recommendation fixes it. Would it be worth adding wcwidth as a dependency?

ianmcorvidae, 2024-12-22:
With #723 this should now be installed when meshtastic is installed as `meshtastic[cli]` (which as of recently also includes some other cosmetic sorts of dependencies, such as `print_color`). It won't be installed with a very basic install, especially so that library users don't need to install the wcwidth library.

closed 2024-12-22 by ianmcorvidae

brad28b, 2024-12-22:
It still seems to be broken with the 👀 character, see attached. This is after a new install of meshtastic[cli], using Windows 11.

[image]

ianmcorvidae, 2024-12-22:
I think you may have tested before I formally cut the release that included this. Could you try updating again to CLI version 2.5.7, with `meshtastic[cli]`? That should install `wcwidth` and display this correctly.

brad28b, 2024-12-22:
I just tried again, those googley eyes emoticon still seem to screw it up, but it works fine for all other emoticons. 

Confirm wcwidth was installed.

PR #723 (2024-12-22): Add 'wcwidth' as an optional cli dependency
Fixes #598 when installed as `meshtastic[cli]`, as should now/soon be recommended by documentation.
commit 22750778 (2024-12-22): Add 'wcwidth' as an optional cli dependency

Fixes #598 when installed as meshtastic[cli], as should now be
recommended by documentation.