meshtastic/python #599: [FYI] /dev/ttyUSB0 vs /dev/ACM0

brad28b, 2024-06-18:
This is mostly an FYI in case it helps others, including developers if this is some kind of bug.

I had one of my RAK4631 nodes connected to an old Raspberry Pi 2  running Debian. And the Meshtastic Python CLI was always really hit and miss. Sometimes it'd work, after a long while, other times, it wouldn't work at all (timeouts).
 
Today I moved my node and plugged it into a different Linux (Ubuntu) box (more modern Intel NUC), and the first difference I noticed was that the meshtastic python CLI never failed on any command, it connected to the radio instantly, and always succeeded without any delay.

The only difference I can see is that on the old Raspberry Pi the RAK4631 was detected as /dev/ttyUSB0, where as on the Intel NUC it's detected as /dev/ACM0

Not sure what the significance of that is. But it works reliably now.

joneyes, 2024-06-18:
Oddly enough I have come across a similar problem today with Lilygo Tlora T3S3 and Heltec Wireless Stick Lite V3.
The heltec flashes using CLI without fail on the Pi and is detected as tty/USB0
It also passes the test with esptool.py chip_id

The T3S3 on the other hand is detected as tty/ACM0 cannot be flashed via CLI unless the boot button is pressed, and that is the only way it will pass the esptool.py chip_id test too (though it doesn't recover from this either without a power cycle.)
I also have a Tbeam Supreme, which is S3 core, and this behaves the same as the T3S3.

Pressing the boot button is a bit of a pain, as the Pi and the T3S3 are in a weatherproof box outside - this is the reason i got the WSL. Unfortunately, the WSL is not as good as the T3S3 and it often fails to receive ack's for Tx packets, I think it's a bit deaf or off frequency....

It doesn't seem to be possible to flash the Lilygo's using flasher.meshtastic page from Chromium on the Pi - i don't think it gets on with the change of state of the port.
Now I can bring the T3S3 inside onto the Windows PC and it will happily flash using flasher.meshtastic using the 1200bd reset first - this changes the USB port address.
The behaviour is the same on a windows machine with CLI on the Lilygo S3 units - the boot button has to be pressed to get them into flash mode. There's obviously a way around it if the web flasher can do it?

jcourter, 2026-05-11:
I just ran into this, so I'm going to add a little more information based on my limited testing.  I can't reproduce this on the ESP32-based nodes I have (Heltec v3 and v4) but I _can_ reproduce this reliably with any of the nrf52840-based devices I have tried (several variations of the DIY nrf52840 board and a Heltec T096).  The easiest test I've been able to do to duplicate the issue is to set up a cron job to send a message once a minute.  After 2 or 3 times, the nrf52840 devices (accessed through /dev/ttyACMx) will start to fail most of the time (there is no error given, it simply does not transmit the message).  Interestingly, the device will sometimes transmit a message right after you disconnect it from USB.  Again, doing this same test on an ESP32-based device (accessed via /dev/ttyUSBx) is 100% reliable.

Edited to add - I don't have any trouble with the nrf52840 boards running things that keep the serial port open the whole time (TC2-Mesh-BBS for example), so it seems to be some trouble with opening/closing the connection.