meshtastic/python #601: Reducing the number of dependencies

njh, 2024-06-20:
I have been looking at packaging the meshtastic CLI, so that it can easily be installed using `brew install meshtastic` on a Mac.
There are a lot of dependencies listed in `requirements.txt`, although many of them are just needed for testing.
In #587, @FedericoCeratto has been looking at reducing the number of dependencies by removing [timeago](https://pypi.org/project/timeago/), which seems like a good idea to me.

[pexpect](https://github.com/pexpect/pexpect) is listed in both `setup.py` and `requirements.txt` and but doesn't seem to be being used anywhere? Maybe I am missing something? Can it be removed?

There two are listed in `requirements.txt` but don't seem to be being used, however I can see why they might be useful:
- `webencodings`
- `pyparsing`

And maybe they are an optional extra for one of the other dependencies?

geeksville, 2024-06-20:
I have a WIP branch that changes over to Poetry which addresses much of this.  I'll try to send in a PR soon.

ianmcorvidae, 2024-06-20:
poetry is definitely the way forward; I think that a more condensed list of dependencies is in `setup.py` (which the poetry migration will replace), if it's helpful to get started on that now for the purposes of packaging. I also just pushed a commit removing `pexpect` from `setup.py` (and one unused import of `dotmap`), which can go out next release.

I'll try to take a look at 587 again soon. I'm not sure Federico is has looked at it in a bit, so I'll maybe try to write some tests for it/try it out a bit so it can get in and remove that `timeago` dependency.

for now that does at least cut the install requirements down to: `pyserial`, `protobuf`, `requests`, `pypubsub`, `dotmap`, `pyqrcode`, `tabulate`, `timeago`, `pyyaml`, `bleak`, and `packaging` (plus, for tunnel, but that's linux-only, `pytap2`). If you're packaging the CLI in brew, you might also include `wcwidth` for #598

njh, 2024-06-21:
That all sounds great, thanks Ian! 👍 

I have a working Homebrew Formula, which passes `brew audit --strict --online meshtastic`:
https://gist.github.com/njh/d4c8a61bdf7b3df7ee6cd0ac4e8432f2

All of the dependencies have a Source Distribution (.tar.gz) on PyPi apart from pypubsub, which I have asked for here:
https://github.com/schollii/pypubsub/issues/48

An optimisation would be to not install `mesh-tunnel` CLI if pytap2 isn't available (or not on Linux).

Oh and it would be good to clarify the license (#582), which I currently have listed as `Apache-2.0`.

geeksville, 2024-06-21:
(I'm just adding myself as an assignee only for the 'make poetry work' portion of this good idea.  making the assignment so I can autokeep this in my neartearm todo list)

geeksville, 2024-06-21:
added an issue for the poetry stuff #604.

ianmcorvidae, 2024-06-26:
I've merged the removal of `timeago` and the change to using `poetry`, both of which will hopefully help with this! I might see if I can sometime soon restructure the one spot that's currently using `dotmap` to get rid of that one too.

As far as licensing, I don't think Apache2 is an inaccurate license to use since we haven't really done a proper formal relicensing, but GPL3 is probably the more future-proof choice. The idea, as I was told, is that this is largely the only Meshtastic thing that's not already GPL3, so the intent has been to move it over to that, which should be a compatible relicensing as far as I can tell (IANAL, of course, and if anyone knows a reason it's not, please do let us know).

I'd love to set it up so `mesh-tunnel` and `pytap2` aren't needed or installed except on Linux -- perhaps poetry has a good way of doing this? I'll need to dig around a bit to get familiar with poetry to be sure, though. I think beyond that we're down to stuff that at least the CLI is definitely using. I would like, potentially, to have a more limited version for library-only users, who probably don't need `pyqrcode`, `tabulate`, and possibly `requests` & `packaging` (which are used for checking if there's a new version of the library/CLI), but that's probably a separate thing.

garthvh, 2024-06-26:
Apache > GPL3 is fine, the other way is potentially an issue.  https://www.apache.org/licenses/GPL-compatibility.html#:~:text=Apache%202%20software%20can%20therefore,be%20included%20in%20Apache%20projects.

alexmyczko, 2024-06-27:
just did it for debian, but also need it with brew on macOS: https://ftp-master.debian.org/new/meshtastic_2.3.12-1.html

alexmyczko, 2024-07-09:
yay https://tracker.debian.org/news/1543170/accepted-meshtastic-2312-1-source-all-into-unstable/

ianmcorvidae, 2024-12-21:
https://github.com/meshtastic/python/pull/711 reduced the required dependencies for this even more, and https://github.com/meshtastic/python/commit/1a5ca789c2ce25d724b83cc74d0916f2b3aa9830 and https://github.com/meshtastic/python/commit/89b41c1a19e4f139bb344173500b9454d67a25cc removed a couple which I'm not sure why they were still present. I'm going to close this for now, but am happy to entertain any potential reductions in dependencies, especially for using this code as a library.

closed 2024-12-21 by ianmcorvidae