meshtastic/python #608: moveing auto generated protobuf glue into meshtastic.generated.

geeksville, 2024-06-22:
Related to #579. What about moving the generated protobuf glue out of the root of the meshtastic project.  Instead have it in meshtastic.generated.  Do ya'll (in particular @ianmcorvidae think that's a good idea)?

If so, I can send in a PR soonish.

It will be a breaking change if there are any external users of our API that have reached in and touched those classes (though I didn't find any in my googling).

(Also I'm sorry I didn't do this originally before there was so many great folks using/improving this - it would have been less painful if they were in the right place all along)

geeksville, 2024-06-24:
@ianmcorvidae any opinion on this?  if you are fine with it I can make a PR.

ianmcorvidae, 2024-06-26:
I think this is a good idea.

I do know there's some folks who are importing the `_pb2` modules directly in order to use them (particularly folks who are pulling stuff straight off MQTT). I think we can just be sure to mark it clearly in the release notes when we do it. It's not a huge change to stick in the extra folder in imports, so it shouldn't be a painful migration for anyone I don't think.

geeksville, 2024-07-03:
done!

closed 2024-07-03 by geeksville