meshtastic/python #612: String concatenation error in traceroute

ianmcorvidae, 2024-06-26:
Got this running a trace the other day:

```
$ meshtastic --host xxxx --ch-index 1 --traceroute '!xxxx'
Connected to radio
Sending traceroute request to !xxxx on channelIndex:1 (this could take a while)
Route traced:
ERROR file:stream_interface.py __reader line:175 Error while handling message from radio can only concatenate str (not "NoneType") to str
Traceback (most recent call last):
  File "/home/mian/Source/meshtastic-python/venv/lib/python3.8/site-packages/meshtastic/stream_interface.py", line 173, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/home/mian/Source/meshtastic-python/venv/lib/python3.8/site-packages/meshtastic/mesh_interface.py", line 888, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/home/mian/Source/meshtastic-python/venv/lib/python3.8/site-packages/meshtastic/mesh_interface.py", line 1148, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/home/mian/Source/meshtastic-python/venv/lib/python3.8/site-packages/meshtastic/mesh_interface.py", line 469, in onResponseTraceRoute
    routeStr += " --> " + self._nodeNumToId(nodeNum)
TypeError: can only concatenate str (not "NoneType") to str
```

filing a bug so I don't forget about it -- I think it's probably a fairly rare bug to come up since it's probably an unknown-node thing.

Talie5in, 2024-06-26:
@ianmcorvidae Firmware 2.3.12 or newer?
If so, may be related to this change https://github.com/meshtastic/firmware/pull/4056

ianmcorvidae, 2024-06-30:
I don't know if there was a 2.3.12 firmware in the mix or not; the relevant function returns `None` any time it doesn't have a node in the nodedb, it appears, so I don't think that's a requirement. It just doesn't come up that often, probably.

PR #621 (2024-06-30): Properly handle missing nodes in traceroute response
Fixes #612
commit 93fbc784 (2024-06-30): Properly handle missing nodes in traceroute response (fixes #612)
closed 2024-06-30 by ianmcorvidae