meshtastic/python #614: Bluetooth device logging interface

thebentern, 2024-06-27:
In firmware version 2.3.14, a new bluetooth characteristic `6c6fd238-78fa-436b-aacf-15c5be1ef2e2` was added to be a notify characteristic for serial style logs from the connected device when `bluetooth.device_logging_enabled` is true.

I've only implemented this on Apple clients so far:
[image]

I think it would be a neat feature to expose on Python as well, perhaps in a similar manner to --noproto, since they're the same logs essentially.

ianmcorvidae, 2024-06-30:
@thebentern is this closed by #609 ? I'm being lazy about reading to be sure if there's any extra work we need to do there.

thebentern, 2024-06-30:
Closed by #616

closed 2024-06-30 by thebentern