meshtastic/python #623: set-ham cannot be reverted

jranma, 2024-06-30:
looks like "set-ham" setting cannot be reverted. There is no way, once set, to unset it using python client. 

meshtastic --set-ham false
Connected to radio
Setting Ham ID to false and turning off encryption
Writing modified channels to device

garthvh, 2024-06-30:
Looks like it set properly, what part did not take?

ianmcorvidae, 2024-06-30:
I think they'd like to turn ham mode _off_, which it's true doesn't look like it's possible very nicely with CLI. However, there's no real way we can guess what encryption keys you'd like each channel to have, so to disable ham mode I think you'd need to set the short & long names (which I think would set the `is_licensed` flag to false) plus the channel settings you prefer (most readily with `--seturl`). Alternately, you could factory reset, if your desired settings are close to the defaults. I'll make a note to check that changing short/long name disables `is_licensed` and document it.