meshtastic/python #630: request  for settings via remote admin returns wrong info

zl3ag, 2024-07-03:
Example 1:

# meshtastic --port /dev/ttyACM0 --dest '!25c2f43c' --get position
Connected to radio
Requesting current config from remote node (this can take a while).

serial:

Completed getting preferences

Example 2:

# meshtastic --port /dev/ttyACM0 --dest '!25c2f43c' --get network
Connected to radio
Requesting current config from remote node (this can take a while).

store_forward:

Completed getting preferences

ianmcorvidae, 2024-07-03:
What version of the CLI are you running? This is an issue that came up at one point the other month, but was fixed then. There was some changes to related code recently though, so it could be a regression.

zl3ag, 2024-07-04:
Upgraded to v2.3.13 just before experimenting with remote admin for the first time.

ianmcorvidae, 2024-07-04:
Ah, yes, I see the issue. Same spot as before, unfortunately. I'll get a fix put in before the next release, which should hopefully be pretty soon.

commit 15b03b70 (2024-07-04): Correctly detect LocalConfig vs. LocalModuleConfig, again. fixes #630
commit 2c8fd8b6 (2024-07-04): Merge pull request #632 from ianmcorvidae/remote-config-wrongtype-again

Correctly detect LocalConfig vs. LocalModuleConfig, again. fixes #630
closed 2024-07-04 by ianmcorvidae