meshtastic/python #634: meshtastic --get returns unrelated config options in remote admin mode

nolith, 2024-07-07:
Hello, I'm setting up a remote node. I'm operating it using the python CLI and a wifi enabled node in my network.

Today I noticed that the get commands returns informations unrelated to the requested config. I.e. quen asking the device info I'm getting the mqtt config.

```
meshtastic --host 10.10.107.3 --dest '!4358e29c' --get device
Connected to radio
Requesting current config from remote node (this can take a while).

mqtt:
address: "mqtt.meshtastic.org"
username: "meshdev"
password: "large4cats"
encryption_enabled: true
root: "msh/EU_868"

Completed getting preferences
```

```
 meshtastic --version
2.3.13
```

Both device are heltec v3 running 2.3.13 fw.

Power-cycling both nodes didn't helped

jranma, 2024-07-07:
I can reproduce this bug. I think it was working fine with python client 2.3.12

nolith, 2024-07-07:
> I can reproduce this bug. I think it was working fine with python client 2.3.12

Thanks for the hint. I downgraded the python CLI to 2.3.12 and it works as intended

jranma, 2024-07-07:
> I can reproduce this bug. I think it was working fine with python client 2.3.12

How du you manage to downgrade? Im using windows

nolith, 2024-07-07:
> > I can reproduce this bug. I think it was working fine with python client 2.3.12
> 
> 
> 
> How du you manage to downgrade? Im using windows

I'm not sure I can help on windows, but I am using a python venv to manage my meshtastic stuff. I have a requirements.txt with all the versions of the tools I'm using. I just reverted it to the old version and run pip install -r requirements.txt

Maybe you can try with pip install meshtastic==2.3.12

ferrellw, 2024-07-08:
I was able to manage local nodes running 2.3.13 with CLI v2.3.13. I was *NOT* able to manage any remote nodes with CLI 2.3.13. Reverting firmware and CLI to 12.3.12 restored remote node management functionality. 

I was receiving timeout errors even tho radios could message each other on the 'admin' channel. I don't know if my findings are helpful but I wanted to share in case it offered any useful information.

```meshtastic --host 192.168.1.141 --dest '!0c18be04' --no-time --get bluetooth.enabled                                                    71.218.142.149   100% 
Connected to radio
Requesting current config from remote node (this can take a while).
Aborting due to: Timed out waiting for an acknowledgment
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 824, in onConnected
    found = getPref(node, pref[0])
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 128, in getPref
    node.requestConfig(config_type)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/node.py", line 140, in requestConfig
    self.iface.waitForAckNak()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 639, in waitForAckNak
    raise MeshInterface.MeshInterfaceError("Timed out waiting for an acknowledgment")
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for an acknowledgment

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 1570, in main
    common()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 1086, in common
    onConnected(client)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 884, in onConnected
    interface.close()  # close the connection now, so that our app exits
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/tcp_interface.py", line 60, in close
    StreamInterface.close(self)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/stream_interface.py", line 120, in close
    MeshInterface.close(self)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 125, in close
    self._sendDisconnect()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 761, in _sendDisconnect
    self._sendToRadio(m)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
```

or

```meshtastic --host 192.168.1.141 --dest '!0c18be04' --set lora.hop_limit 4 --get lora                                                    71.218.142.149   100% 
Connected to radio
Requesting current config from remote node (this can take a while).
Aborting due to: Timed out waiting for an acknowledgment
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 524, in onConnected
    node.requestConfig(
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/node.py", line 140, in requestConfig
    self.iface.waitForAckNak()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 639, in waitForAckNak
    raise MeshInterface.MeshInterfaceError("Timed out waiting for an acknowledgment")
meshtastic.mesh_interface.MeshInterface.MeshInterfaceError: Timed out waiting for an acknowledgment

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/meshtastic", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 1570, in main
    common()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 1086, in common
    onConnected(client)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/__main__.py", line 884, in onConnected
    interface.close()  # close the connection now, so that our app exits
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/tcp_interface.py", line 60, in close
    StreamInterface.close(self)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/stream_interface.py", line 120, in close
    MeshInterface.close(self)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 125, in close
    self._sendDisconnect()
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 761, in _sendDisconnect
    self._sendToRadio(m)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/usr/local/lib/python3.10/dist-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
```

ianmcorvidae, 2024-07-08:
This main issue is the same issue as #630 -- already fixed in dev. The timeouts will be a separate thing. Closing this for now; I'll try to make a release soon to get this out as 2.3.14.

closed 2024-07-08 by ianmcorvidae