meshtastic/python #635: get network vomits

zl3ag, 2024-07-08:
Running latest 2.3.14 version (thanks for fixing that other problem in .13 BTW!

meshtastic --port /dev/ttyACM0 --dest '!25c2f43c' --get network
Connected to radio
Requesting current config from remote node (this can take a while).

ERROR file:stream_interface.py __reader line:175 Error while handling message from radio Assignment not allowed to message, map, or repeated field "ipv4_config" in protocol message object.
Traceback (most recent call last):
  File "/root/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 173, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/root/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 926, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/root/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1186, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/root/.local/pipx/venvs/meshtastic/lib/python3.11/site-packages/meshtastic/node.py", line 119, in onResponseRequestSettings
    setattr(config_values, camel_to_snake(key), value)
AttributeError: Assignment not allowed to message, map, or repeated field "ipv4_config" in protocol message object.
Completed getting preferences

ianmcorvidae, 2024-07-08:
Ah, looks like there's some issues with the nested field there; I suspect without stuff in `ipv4_config` it still works. We should figure out how to extract that information from the protobuf classes to handle this sort of thing automatically; hopefully that's possible.

ianmcorvidae, 2024-09-12:
It will need testing, but this is probably fixed with 2.5 (which I am about to go click the button to release), made some changes to how this handler works.

ianmcorvidae, 2024-12-21:
I'm going to assume this is fixed with 2.5. Please let me know if not.

closed 2024-12-21 by ianmcorvidae