meshtastic/python #641: Heltec V3 with ESP32 S3 works over CP210x but not over USB JTAG.

MichaelPWalter, 2024-07-23:
The Heltec V3 comes with the ESP32 S3 which does not seem to be explicitly supported yet. Communication over the USB port utilizes the CP210x so it works. The S3 specifically has an additional USB JTAG interface which is exposed over the GPIO headers so connection would be more convenient. 
D+ is located on GPIO Pin 20 and D- on GPIO Pin 19. Power HAS to be provided over the 5v rail, not Ve. 

Is there a way to establish a solid connection via the GPIO pins ? I do sometimes get a response using meshtastic --noproto but after the initial header it's always completely silent. Let me know what i can do to help.

MichaelPWalter, 2024-07-23:
I've analyzed the chip in accordance to /info/ubuntu/diy.txt as follows. This can be seen as /info/ubuntu/heltec_v3.txt

`lsusb 

Bus 001 Device 009: ID 303a:1001 Espressif USB JTAG/serial debug unit
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb -d 303a: -v

Bus 001 Device 009: ID 303a:1001 Espressif USB JTAG/serial debug unit
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x303a
  idProduct          0x1001
  bcdDevice            1.01
  iManufacturer           1 Espressif
  iProduct                2 USB JTAG/serial debug unit
  iSerial                 3 48:CA:43:59:06:20
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0062
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0
      iFunction               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0
      iInterface              0
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      2
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      1
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

esptool.py chip_id

esptool.py v4.7.0
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded Flash 8MB (GD)
Crystal is 40MHz
MAC: 48:ca:43:59:06:20
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S3 has no Chip ID. Reading MAC instead.
MAC: 48:ca:43:59:06:20
Hard resetting via RTS pin...

`

Abdull, 2024-09-28:
did you solder those empty resistor pads on the PCB in order to have the ESP32-S3's USB D+ and D- package leads connected to the header pins? Because at least for the Heltev LoRa V3, V3.1, and V3.2 hardware revisions, this must be done first - and probably the resistors that hook up the USB-C port's D+ and D- to the CP2102 USB UART chip have to be removed, otherwise there would be three USB devices (the ESP32-S3, the CP2102, and the USB-C connected devices) competing for a bus ressource that can only handle exactly two.

MichaelPWalter, 2024-09-29:
I will have to try that :)

Abdull, 2024-10-01:
via https://resource.heltec.cn/download/WiFi_LoRa_32_V3/HTIT-WB32LA_V3(Rev1.1).pdf : 
[image]

----

via https://resource.heltec.cn/download/WiFi_LoRa_32_V3/HTIT-WB32LA(F)_V3.1_Schematic_Diagram.pdf :

[image]

Note: don't confuse the different pins and signals labeled with "19" and "20": 19 and 20 are reused for IC-specific pins as well.

----

via https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-guides/jtag-debugging/configure-builtin-jtag.html : 

[image]

----

ESP32-S3 datasheet (includes further information about steps required to set up USB JTAG): https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf :

ianmcorvidae, 2024-12-21:
It sounds like this may not be an issue with the CLI but with your setup. I'll close this for now, but let me know if there's more that might be on the CLI end.

closed 2024-12-21 by ianmcorvidae